zlgopen/awtk #735: 请问如何把控制的value值传回嵌入式系统中使用

skdswk, 2022-05-30:


skdswk, 2022-05-30:
我是说我HMI中有一个面页通过progress bar设置亮度值， 要把这个设置的值传出来，用于ROTS去设置实际的背光亮度。 但这个progress bar的value值，我取不出来，直接赋给一个全局变量没作用、

skdswk, 2022-05-30:
int32_t value = (PROGRESS_BAR(progress_bar)->value + delta);
  if (value >= 100)
  {
    value = 100;
  }
  if (value <= 0)
  {
    value = 0;
  }
  widget_animate_value_to(progress_bar, value, 0);
  
###   CAR_INFO_SETTING.setting_LCD_background = value;
但实际上我在设置端口值时，取CAR_INFO_SETTING.setting_LCD_background值，取不到

xianjimli, 2022-05-30:
应该用slider控件吧。注册EVT_VALUE_CHANGED或者EVT_VALUE_CHANGING事件，在处理函数里设置背光。

```
 widget_t* slider = slider_create(win, 10, 10, 200, 30);
 widget_on(slider, EVT_VALUE_CHANGED, on_changed, NULL);
 widget_on(slider, EVT_VALUE_CHANGING, on_changing, NULL);
```

skdswk, 2022-05-30:
你这方法是控制slider的bar显示的方法，还是外界的值变化了slider的进度跟着变。 
我现在的需求是，根据slider的值来设置背光变化，要获取slider值赋给外面的全局变量，你试着这个赋值赋不了。

skdswk, 2022-06-03:
解决了，赋值是成功的，其他地方函数做错了

closed 2022-06-03 by skdswk