zlgopen/awtk #745: 移植到嵌入式linux系统中，如何通过gpio按键来触发事件，完成指定操作

liuguohuan-xiezuo, 2022-07-14:
在window上编写的awtk，可以通过键盘鼠标触发button，拖动，滚轮来实现各种事件触发，如EVT_CLICK，那移植到开发板上，只有gpio按键，高低电平，怎么去完成各种事件触发呢，例如页面跳转之类操作，能通过什么方式去实现呢

xianjimli, 2022-07-14:
转换成按键事件就好了。

liuguohuan-xiezuo, 2022-07-14:
具体需要怎么操作呢，是把控件转换成按键事件吗？gpio又怎么对应是哪个按键按下呢

liuguohuan-xiezuo, 2022-07-15:
> 转换成按键事件就好了。

具体需要怎么操作呢，是把控件转换成按键事件吗？gpio又怎么对应是哪个按键按下呢，例如我有4个gpio按键，想定义为确认，取消，上，下，我该怎么对应按键跟事件的关系呢？

xianjimli, 2022-07-15:
参考一下这个：https://github.com/zlgopen/awtk-stm32f103ze-raw/blob/master/awtk-port/main_loop_stm32_raw.c

liuguohuan-xiezuo, 2022-07-18:
> 参考一下这个：https://github.com/zlgopen/awtk-stm32f103ze-raw/blob/master/awtk-port/main_loop_stm32_raw.c
因为不太熟悉stm32，看了这个例子之后好像也不怎么理解，看这个项目大概的意思是初始化，时钟，中断，gpio，然后就gpio按下，把一个值传到main_loop_post_pointer_event()函数，这个函数是用来干嘛的呢，最终怎么转换到ui上去出发事件呢

xianjimli, 2022-07-20:
需要了解内部流程，可以看代码，用调试器调试一下。