zlgopen/awtk #748: svg图片的使用问题

WeSrc-Yin, 2022-07-26:
李老师，你好，关于svg图片的使用我遇到以下问题，麻烦指导回复下：
1.svg图片要怎么缩放？我使用widget_move_resize和widget_resize都不能改变图片的大小；
2.svg图片制作有什么需求吗？比如，我现在知道awtk不支持svg图片透明色。但我们美工做出来的svg图片显示出来是黑色，我使用awtk demo里面的svg图片（比如china这张图片）是可以显示出来的，帮我看下我的xml代码和svg图片有啥问题？（我在下面上传svg图片和xml代码）

WeSrc-Yin, 2022-07-26:
[image]

WeSrc-Yin, 2022-07-26:
我的xml代码如下，因为我要旋转，所以使用了guage_pointer控件里面套svg图片
`<window closable="no" text="AVP" anim_hint="htranslate" single_instance="true" theme="default" style="test">
   <gauge_pointer name="china" x="0" y="0" w="200" h="200" value="0" image="china"/>
   <gauge_pointer name="car" x="0" y="b" w="24" h="55" value="0" image="car2"/>
</window>`

WNsACE, 2022-07-27:
1，使用 svg 的控件也可以旋转，demouiold 里面有 svg 的例子，你可以看看
2，awtk 的 svg 现在暂时支持画点画线填充而已，还不支持 svg 特有的叠加消除的效果。

WeSrc-Yin, 2022-07-27:
> 1，使用 svg 的控件也可以旋转，demouiold 里面有 svg 的例子，你可以看看 2，awtk 的 svg 现在暂时支持画点画线填充而已，还不支持 svg 特有的叠加消除的效果。

1.demouiold里面的例子是使用了动画，我的需求是：根据外部输入值修改svg的位置及旋转角度以及缩放；能不能告诉我使用awtk哪个api接口可以实现svg图片的旋转、位置改变、缩放？

2.我的svg图片也很简单，如上图，为什么我的svg图片显示成黑色的了？

WNsACE, 2022-07-27:
1，通过 widget_set_prop 来设置 scale_x/scale_y 可以缩放，设置 rotation 可以旋转，设置 anchor_x/anchor_y 可以设置旋转锚点
2，应该这个 svg 里面使用了 style 来设置填充颜色，awtk 支持解析这个 style 标签，所以导致颜色没有了。

WeSrc-Yin, 2022-07-27:
搞定了，谢谢。

WeSrc-Yin, 2022-07-27:
> 1，通过 widget_set_prop 来设置 scale_x/scale_y 可以缩放，设置 rotation 可以旋转，设置 anchor_x/anchor_y 可以设置旋转锚点 2，应该这个 svg 里面使用了 style 来设置填充颜色，awtk 支持解析这个 style 标签，所以导致颜色没有了。

1.搞定了，谢谢；
2.能明确一下awtk所支持的svg的要求吗，或者说awtk demo里面的svg图片是用用什么工具做的，制作过程有什么要求和需要注意的地方？现在就是我们美工做出来的svg图片使用awtk显示是黑色的，但浏览器打开是正常的。

WNsACE, 2022-07-28:
1，都可以，0.0-1.0 是百分比，后面加 px 的话，就变成像素了，锚点是基于控件来说的
2，这个没有明确的说明文档，是因为路径的颜色没了，所以就全部都变成了黑色了，还有需要注意的是，awtk 不支持 svg 的叠加消除功能，

closed 2022-08-03 by WeSrc-Yin