zlgopen/awtk #757: windows系统下conf_json模块的`conf_json_load`函数总是失败

ufbycd, 2022-08-22:
经测试，`conf_json_load`函数在linux下没问题而在windows下总失败。
大致查了下，似乎是`data_reader_factory_create_reader`函数对文件路径解析不正确导致，如果文件路径是包含盘符如` D:\projects\host\release\test.json`路径解析就不正确。

必须要将文件路径转成 `file://`这种格式吗？

ufbycd, 2022-08-22:
尝试按以下方式修改就成功了：即判断下`:`之后是否是`\`，如果是那这个路径应该是windows下的包含盘符的路径

```diff
diff --git a/src/tkc/data_reader_factory.c b/src/tkc/data_reader_factory.c
index 1470dd29d..6ed984db9 100644
--- a/src/tkc/data_reader_factory.c
+++ b/src/tkc/data_reader_factory.c
@@ -87,7 +87,7 @@ data_reader_t* data_reader_factory_create_reader(data_reader_factory_t* factory,
   p = strchr(url, ':');
   memset(protocol, 0x00, sizeof(protocol));

-  if (p != NULL) {
+  if ((p != NULL) && (p[1] != '\\')) {
     return_value_if_fail((p - url) < TK_NAME_LEN, NULL);
     tk_strncpy(protocol, url, p - url);
   } else {
```

xianjimli, 2022-08-30:
好的，谢谢。改了

closed 2022-08-30 by ufbycd