zlgopen/awtk #758: mledit显示utf-8编码文字出现崩溃的问题

hanzixi, 2022-08-24:
使用widget_set_text_utf8接口给mledit控件设置显示的文本,测试发现设置的文本中如果某一个字节的最高位位1就会出现崩溃的问题,比如汉字"了"的utf8编码为0xE4BA86,设置显示之后程序就直接退出了,出现了以下错误"build/var/awtk/3rd/libunibreak/wordbreak.c:130:set_brks_to: Assertion `ch != EOS' failed.Aborted";
然后特别测试了下其中一字节最高位为1的情况,比如显示字符串30 31 32 33 34 FF 36(十六进制)就会出现类似的崩溃问题,错误如下:" build/var/awtk/3rd/libunibreak/linebreak.c:799: set_linebreaks: Assertion `posLast == posCur - 1 && posCur <= len' failed.
Aborted";把0xFF修改为0x1F就能正常,不出现崩溃的问题;
同样的方法测试edit显示文本不会出现类似的问题;推测好像是计算换行出现的问题,烦请帮忙看下问题,谢谢;

hanzixi, 2022-08-24:
使用的sdk版本是V1.7,设备是嵌入式Linux,cortex-A7核心的cpu;

xianjimli, 2022-08-26:
看看这个文档：https://developer.zlg.cn/zStation/articleinfo?id=852755

hanzixi, 2022-08-26:
谢谢李先生的解答  已经完美解决该问题   排查下缺失是编码的问题   使用串口工具发送的汉字默认是gbk编码   我修改编码格式之后以为生效了   起始并没有生效  还是发送的gbk编码   换一个串口工具就可以了   再次感谢;

closed 2022-08-26 by hanzixi