zlgopen/awtk #762: 如何监听操作系统发给窗口的事件？

ufbycd, 2022-08-31:
最近在做USB设备的热插拔识别功能，发现libusb在Windows系统上不支持热插拔，原因大概是Windows的热插拔事件只发给UI窗口。
那么只能自己去监听操作系统发给窗口的插拔事件来实现插拔识别了。AWTK目前可以监听操作系统事件么？
Qt上可以先通过`void QCoreApplication::installNativeEventFilter(QAbstractNativeEventFilter *filterObj)`安装native事件过滤器然后在主窗口连接`MainWindow::NativeDeviceChanged`信号即可。

ufbycd, 2022-08-31:
看了下源码，AWTK那边需要使能SDL2的`SDL_SYSWMEVENT`事件，并在主循环中处理并分发此事件，然后像Qt那样对窗口句柄添加插拔事件过滤器。

xianjimli, 2022-08-31:
好的，我抽空看看。

xianjimli, 2022-09-01:
我增加system_event_t(EVT_SYSTEM)，通过窗口管理器注册。但是我插拔USB并没有收到该事件，你有空帮我试试。谢谢

ufbycd, 2022-09-02:
需要调用win32 API `RegisterDeviceNotification`对窗口注册插拔事件过滤器才会收到插拔事件，该函数的第一个参数为窗口句柄。
我在AWTK之外用如下代码进行注册，经测试是可以收到插拔事件的，可参考下： 其中`hWnd`为当前的窗口句柄
```C
static const GUID GUID_DEVINTERFACE_LIST[] =
{
    // GUID_DEVINTERFACE_USB_DEVICE
    {0xA5DCBF10, 0x6530, 0x11D2, {0x90, 0x1F, 0x00, 0xC0, 0x4F, 0xB9, 0x51, 0xED}},
    // GUID_DEVINTERFACE_DISK
    {0x53f56307, 0xb6bf, 0x11d0, {0x94, 0xf2, 0x00, 0xa0, 0xc9, 0x1e, 0xfb, 0x8b}},
    // GUID_DEVINTERFACE_HID,
    {0x4D1E55B2, 0xF16F, 0x11CF, {0x88, 0xCB, 0x00, 0x11, 0x11, 0x00, 0x00, 0x30}},
    // GUID_NDIS_LAN_CLASS
    {0xad498944, 0x762f, 0x11d0, {0x8d, 0xcb, 0x00, 0xc0, 0x4f, 0xc3, 0x35, 0x8c}}
    //GUID_DEVINTERFACE_COMPORT
    //{ 0x86e0d1e0, 0x8089, 0x11d0, { 0x9c, 0xe4, 0x08, 0x00, 0x3e, 0x30, 0x1f, 0x73 } },
    //GUID_DEVINTERFACE_SERENUM_BUS_ENUMERATOR
    //{ 0x4D36E978, 0xE325, 0x11CE, { 0xBF, 0xC1, 0x08, 0x00, 0x2B, 0xE1, 0x03, 0x18 } },
    //GUID_DEVINTERFACE_PARALLEL
    //{ 0x97F76EF0, 0xF883, 0x11D0, { 0xAF, 0x1F, 0x00, 0x00, 0xF8, 0x00, 0x84, 0x5C } },
    //GUID_DEVINTERFACE_PARCLASS
    //{ 0x811FC6A5, 0xF728, 0x11D0, { 0xA5, 0x37, 0x00, 0x00, 0xF8, 0x75, 0x3E, 0xD1 } }
};

void RegisterDeviceNotify(HWND hWnd)
{
    HDEVNOTIFY hDevNotify;

    for(int i = 0; i < sizeof(GUID_DEVINTERFACE_LIST) / sizeof(GUID); i++)
    {
        DEV_BROADCAST_DEVICEINTERFACE NotificationFilter;
        ZeroMemory(&NotificationFilter, sizeof(NotificationFilter));
        NotificationFilter.dbcc_size = sizeof(DEV_BROADCAST_DEVICEINTERFACE);
        NotificationFilter.dbcc_devicetype = DBT_DEVTYP_DEVICEINTERFACE;
        NotificationFilter.dbcc_classguid = GUID_DEVINTERFACE_LIST[i];
        hDevNotify = RegisterDeviceNotification(hWnd, &NotificationFilter, DEVICE_NOTIFY_WINDOW_HANDLE);
    }
}
```