zlgopen/awtk #768: 放置在system_bar上的popup菜单如果self_layout使用menu布局，在打开后不能显示

ufbycd, 2022-09-27:
system_bar在显示区域顶部，描述如下 ：
```xml
<system_bar h="48">
  <digit_clock x="c" y="m" w="200" h="100%" style="time" format="YYYY-MM-DD hh:mm" />
  <view x="right:50" y="m" w="200" h="100%" children_layout="default(r=1,c=0,s=2)">
    <label w="100" name="user_name" text="admin" />
    <button w="32" text="menu" on:click="open('menu_down_right')" />
  </view>
</system_bar>
```
popup菜单`menu_down_right.xml`描述如下：
```xml
<popup anim_hint="popdown" style="dark" self_layout="menu(position=down,x=right,w=128,h=90)" close_when_click="true">
     <button text="Basic" focusable="true" name="open:basic" />
    <button text="Button" focusable="true" name="open:button" />
    <button name="open:confirm" focusable="true" text="Confirm" />
    <button text="Close" focusable="true" name="close" />
</popup>
```
点击`menu`按钮后菜单没有显示。
改变菜单的布局为如下后，才可以显示：
```xml
<popup x="right:12" y="49" w="128" h="90" anim_hint="popdown" close_when_click="true">
     <button text="Basic" focusable="true" name="open:basic" />
    <button text="Button" focusable="true" name="open:button" />
    <button name="open:confirm" focusable="true" text="Confirm" />
    <button text="Close" focusable="true" name="close" />
</popup>
```

WNsACE, 2022-09-29:
你说的菜单没显示，应该该是出来断言了吧？

ufbycd, 2022-09-29:
> 

没有断言，log信息显示打开了菜单了。

WNsACE, 2022-09-29:
恩恩，刚刚复现问题了，