zlgopen/awtk #780: 嵌入式drm显示，canvas无法刷新成透明

Sokyx, 2022-10-26:
xml如下：
<window theme="stair_home_page" anim_hint="htranslate" >
	<canvas name="preview_canvas" x="0" y="0" w="100%" h="900" />
</window>
代码如下：
static ret_t _home_unlock_on_paint_event(void* ctx, event_t* evt)
{
	widget_t* canvas_widget = WIDGET(ctx);
	canvas_t* canvas = widget_get_canvas(canvas_widget);
	point_t p = {0};
	
	
	p.x = canvas_widget->x;
	p.y = canvas_widget->y;
	widget_to_local(canvas_widget,&p);	//transform to widget local xy
	int w = canvas_widget->w;
	int h = canvas_widget->h;

	canvas_set_fill_color(canvas, color_init(0x00, 0x00, 0xff, 0x00));
	canvas_clear_rect(canvas, p.x, p.y, w, h);
	
	return RET_OK;
}
       以上代码在fb显示下，是可以将canvas区域刷新成全透明，但是在/dev/dri/card0显示下，透明度没有效果，显示的是蓝色。如果在window后面加一个style:bg_color="#ff000000"，好像是会显示透明，但同样在canvas后面加style:bg_color="#ff000000"仍然显示红色，请问这个是啥情况啊？

Sokyx, 2022-10-26:
怎么xml没显示出来：
xml如下：
<window theme="stair_home_page" anim_hint="htranslate" >
	<canvas name="preview_canvas" x="0" y="0" w="100%" h="900" />
</window>

Sokyx, 2022-10-26:
<window theme="stair_home_page" anim_hint="htranslate" >
		<canvas name="preview_canvas" x="0" y="0" w="100%" h="900" />
《/window》

Sokyx, 2022-10-26:
《window theme="stair_home_page" anim_hint="htranslate" 》
				《canvas name="preview_canvas" x="0" y="0" w="100%" h="900" style:bg_color="#ff000000"/》
《/window》

RainMic1019, 2022-10-27:
需要定义宏 WITH_LCD_CLEAR_ALPHA

Sokyx, 2022-10-27:
> 需要定义宏 WITH_LCD_CLEAR_ALPHA

awtk的awtk_config.py和awtk-linux-fb的awtk_config.py里，我都加了-DWITH_LCD_CLEAR_ALPHA=1了，仍然没有效果。而且以前也没加这个，代码里直接填充透明都可以直接把canvas区域刷新透明。现在用的/dev/dri/card0节点，透明没效果了，但是window的style里设置了bg_color="#0000ff00"却是透明的。

Sokyx, 2022-10-27:
> 需要定义宏 WITH_LCD_CLEAR_ALPHA

代码中调用widget_set_opacity设置某个控件透明度都是有效果的，但是我现在是想把指定区域给刷新成透明，能看到下层的视频层，想把canvas所在范围整体透下去，却没有效果

Sokyx, 2022-10-27:
> 你看一下这个文档：https://github.com/zlgopen/awtk/blob/master/docs/how_to_use_mutable_image.md

这个有局限性啊，而且我们播放视频是另一个进程播放的，UI拿不到数据的，只要能刷新透明就行。我看了代码fill_image.inc里面的clear_image函数，rgba的值都是对的，bpp=4，执行tk_memset32((uint32_t*)p, *(uint32_t*)(&data), w);不知道为啥透明度没有起到效果

RainMic1019, 2022-10-27:
你这种只能做图层融合，一个图层刷视频、另一个图层刷ui，然后两个叠加起来显示，只有一个图层透不下去的。

Sokyx, 2022-10-27:
> 你这种只能做图层融合，一个图层刷视频、另一个图层刷ui，然后两个叠加起来显示，只有一个图层透不下去的。

是这样的，我们是有视频层和UI层的，想把视频层放在下方，UI层在上，这样UI上刷新透明就能看到视频了。之前一个平台就是这样处理的，代码也一样，能够把指定区域刷新成透明。现在不知道为啥执行同样的函数，就完全不是透明的。

RainMic1019, 2022-10-27:
这个得看平台是否支持硬件图层融合，如果平台本身不支持的话，是没办法实现的。

Sokyx, 2022-10-27:
> 这个得看平台是否支持硬件图层融合，如果平台本身不支持的话，是没办法实现的。

那先不管图层是否支持融合，不管视频层，现在就在UI层这样去刷新透明，没有效果，这个应该是哪里出了问题，应该跟硬件融合没有关系吧。我没找到是啥问题，但是调用widget_set_opacity设置某个控件透明度都是有效果的，以及设置window的bgcolor透明也是有效果的

RainMic1019, 2022-10-27:
这个跟widget_set_opacity的概念不一样，在窗口半透且没有底色的情况下绘制是会有残留的，你首先确保LCD格式为32位色的，其次定义宏 WITH_LCD_CLEAR_ALPHA 就ok了，我下午测试都是正常的。

Sokyx, 2022-10-27:
> 这个跟widget_set_opacity的概念不一样，在窗口半透且没有底色的情况下绘制是会有残留的，你首先确保LCD格式为32位色的，其次定义宏 WITH_LCD_CLEAR_ALPHA 就ok了，我下午测试都是正常的。

WITH_LCD_CLEAR_ALPHA这个是有定义的，我看代码里的#if WITH_LCD_CLEAR_ALPHA里面都有执行。LCD这个BGRA8888应该也没问题，我看所有颜色和图标显示的都是正常的。

Sokyx, 2022-10-27:
> 这个跟widget_set_opacity的概念不一样，在窗口半透且没有底色的情况下绘制是会有残留的，你首先确保LCD格式为32位色的，其次定义宏 WITH_LCD_CLEAR_ALPHA 就ok了，我下午测试都是正常的。

我设置的填充色是#00ff0000，虽然还是显示的绿色，但是再打开一个半透明的window，看到下方的颜色变成全黑了(正常下方假设是灰色，半透看下去还是灰色，但是这时候下方的绿色变成全黑，像是全透明了一样)。是不是意思我那个绿色已经算是透明了，只不过是残留绿色？

RainMic1019, 2022-10-27:
是的，定义 WITH_LCD_CLEAR_ALPHA 后，在LCD绘制前，会刷上一层透明的颜色，也就是 #00000000，这个在 LCD 上显示本来就是黑色，所以下方的绿色会显示成黑色。

Sokyx, 2022-10-27:
> 是的，定义 WITH_LCD_CLEAR_ALPHA 后，在LCD绘制前，会刷上一层透明的颜色，也就是 #00000000，这个在 LCD 上显示本来就是黑色，所以下方的绿色会显示成黑色。

好的，我明天确认一下，谢谢。

RainMic1019, 2022-10-27:
不客气，如果还有问题，可以加小Z微信（zlgmcu-888）进入AWTK群聊，里面有ZLG的研发工程师。