zlgopen/awtk #786: 我用在windows用designer开发的程序 我怎么在Ubuntu中运行起来呢 scons编译失败

wyztxzs, 2022-11-16:
scons: Reading SConscript files ...
APP_SCRIPTS_ROOT:/home/wyz/my-examples/AwtkApplication5/scripts
AWTK_ROOT: /home/wyz/awtk
AWTK_SCRIPTS_ROOT: /home/wyz/awtk/scripts
MACH=x86_64 ARCH=('64bit', '') TARGET_ARCH=
TK_ROOT: /home/wyz/awtk
awtk_global
fscript_ext_widgets
extwidgets
nfd
widgets
base
gpinyin
fribidi
linebreak
svgtiny
debugger
fscript_ext
streams
conf_io
hal
xml
charset
csv
ubjson
compressors
miniz
tkc_core
mbedtls
/home/wyz/my-examples/AwtkApplication5/bin exist.
/home/wyz/my-examples/AwtkApplication5/lib exist.
False
(u'1280', u'800', u'orentation_0')
AWTK_ROOT: /home/wyz/awtk
TKC_ONLY: False
{}
cd 3rd/awtk-widget-move-circle && scons AWTK_ROOT=/home/wyz/awtk
scons: Reading SConscript files ...
APP_SCRIPTS_ROOT:/home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/scripts
AWTK_ROOT: /home/wyz/awtk
AWTK_SCRIPTS_ROOT: /home/wyz/awtk/scripts
MACH=x86_64 ARCH=('64bit', '') TARGET_ARCH=
TK_ROOT: /home/wyz/awtk
awtk_global
fscript_ext_widgets
extwidgets
nfd
widgets
base
gpinyin
fribidi
linebreak
svgtiny
debugger
fscript_ext
streams
conf_io
hal
xml
charset
csv
ubjson
compressors
miniz
tkc_core
mbedtls
/home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/bin exist.
/home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/lib exist.
False
(u'800', u'480', u'orentation_0')
AWTK_ROOT: /home/wyz/awtk
TKC_ONLY: False
{'AWTK_ROOT': '/home/wyz/awtk'}
src/move_circle.def
src [ 'src' ]
src
Process: /home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/src/move_circle_register.h
Process: /home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/src/move_circle/move_circle.h
fs.js:646
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: EACCES: permission denied, open 'idl/idl.json'
    at Object.fs.openSync (fs.js:646:18)
    at Object.fs.writeFileSync (fs.js:1299:33)
    at IDLGen.saveResult (/home/wyz/awtk/tools/idl_gen/idl_gen.js:458:8)
    at Function.gen (/home/wyz/awtk/tools/idl_gen/idl_gen.js:502:9)
    at Object.<anonymous> (/home/wyz/awtk/tools/idl_gen/index.js:24:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
exe cmd: node "/home/wyz/awtk/tools/idl_gen/index.js" idl/idl.json src "/home/wyz/awtk/tools/idl_gen/idl.json;"  failed.
src/move_circle.def move_circle
fs.js:646
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: EACCES: permission denied, open 'src/move_circle.def'
    at Object.fs.openSync (fs.js:646:18)
    at Object.fs.writeFileSync (fs.js:1299:33)
    at DefGenerator.writeResult (/home/wyz/awtk/tools/dll_def_gen/def_generator.js:36:8)
    at DefGenerator.genJsonAll (/home/wyz/awtk/tools/dll_def_gen/def_generator.js:53:10)
    at DefGenerator.genAll (/home/wyz/awtk/tools/dll_def_gen/def_generator.js:57:10)
    at Function.gen (/home/wyz/awtk/tools/dll_def_gen/def_generator.js:64:9)
    at Object.<anonymous> (/home/wyz/awtk/tools/dll_def_gen/index.js:22:14)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
exe cmd: node "/home/wyz/awtk/tools/dll_def_gen/index.js" idl/idl.json src/move_circle.def failed.
IOError: [Errno 13] Permission denied: '/home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/bin/libawtk.so':
  File "/home/wyz/my-examples/AwtkApplication5/3rd/awtk-widget-move-circle/SConstruct", line 5:
    helper.set_dll_def('src/move_circle.def').set_libs(['move_circle']).call(DefaultEnvironment)
  File "/home/wyz/awtk/scripts/app_helper_base.py", line 637:
    self.prepare()
  File "/home/wyz/awtk/scripts/app_helper_base.py", line 476:
    self.copyAwtkSharedLib()
  File "/home/wyz/awtk/scripts/app_helper_base.py", line 311:
    self.awtk.copySharedLib(self.AWTK_ROOT, self.APP_BIN_DIR, 'awtk')
  File "/home/wyz/awtk/awtk_config_common.py", line 215:
    shutil.copy(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 139:
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 97:
    with open(dst, 'wb') as fdst:

wyztxzs, 2022-11-16:
如题

xianjimli, 2022-11-16:
看下README，安装node依赖的库。

scp1011, 2022-11-19:
项目复制过去的时候".sconsign.dblite"文件要删掉

bjNong, 2022-11-23:
mark