zlgopen/awtk #805: slide_indicator放在list_view_h里面，使用v-for-items自动生成，最大值获取错误的问题

lzh545, 2023-01-11:
xml如下：
<window name="brand_select/brand_select" style:normal:bg_color="#0E1817" v-model="brand_select" single_instance="true">
  <list_view_h name="list_view_h1" x="0" y="0" w="100%" h="368" item_width="200" style:normal:border_color="#00000000" style:normal:bg_color="#303938">
    <scroll_view name="scroll_view" x="1" y="1" w="100%" h="358" children_layout="list_view(item_height=81,cols=5,spacing=8,hl=true)" snap_to_page="true" v-for-items="true">
      <button name="brand_select_button" v-data:text="{item.name}" v-on:click="{return, CloseWindow=true}" text="xx"/>
    </scroll_view>
    <slide_indicator name="slide_indicator" x="0" y="b" w="100%" h="14" max="2" anchor_x="0.5" anchor_y="0" size="14" spacing="21" default_paint="auto"/>
  </list_view_h>
</window>

------------------------------------------------------
简单查了下原因，slide_indicator.c里面，第979行，创建的时候调用了slide_indicator_set_indicated_widget()，没有清slide_indicator->indicated_target，导致slide_indicator_on_layout_children()的时候无法调用slide_indicator_set_indicated_widget()，

[image]
改成
[image]
会不会有问题？

RainMic1019, 2023-01-11:
你是怎么获取最大值的呢？我用 v-for 的用法试了一下，没能重现你说的问题，xml 如下:

<window v-model="home_page_view_model" name="home_page">
  <list_view_h x="center" y="0" w="90%" h="90%" item_width="200" spacing="5">
    <scroll_view name="test" w="100%" h="-20" children_layout="list_view(item_height=120,cols=3,spacing=2,hl=true)" snap_to_page="true" move_to_page="true">
      <view v-for="{items}">
        <button x="0" y="0" w="100%" h="100%" text="1"/>
      </view>
    </scroll_view>
    <slide_indicator x="0" y="b" w="100%" h="20" transition="true" style="blue"/>
  </list_view_h>
  <button name="button" x="127" y="442" w="100" h="36" v-on:click="{add_item}" text="add"/>
</window>

RainMic1019, 2023-01-11:
```
<window v-model="home_page_view_model" name="home_page">
  <list_view_h x="center" y="0" w="90%" h="90%" item_width="200" spacing="5">
    <scroll_view name="test" w="100%" h="-20" children_layout="list_view(item_height=120,cols=3,spacing=2,hl=true)" snap_to_page="true" move_to_page="true">
      <view v-for="{items}">
        <button x="0" y="0" w="100%" h="100%" text="1"/>
      </view>
    </scroll_view>
    <slide_indicator x="0" y="b" w="100%" h="20" transition="true" style="blue"/>
  </list_view_h>
  <button name="button" x="127" y="442" w="100" h="36" v-on:click="{add_item}" text="add"/>
</window>
```

RainMic1019, 2023-01-12:
我这边按照你的说法用 v-for 和 v-for-items 都试了一下，slide_indicator的 小圆点个数都是正常的，建议你先改成 v-for 用法，v-for-items 我们已经不再维护了，后面可能会考虑删掉。

如果使用 v-for 还是有问题的话，可以发个demo我们这边调一下看看。