zlgopen/awtk #810: idl_gen.js parseMethod properties undefined

AzAlam1, 2023-02-07:
`TypeError: Cannot read properties of undefined (reading 'push')
    at IDLGen.parseMethod (/home/awtk/awtk/tools/idl_gen/idl_gen.js:166:24)
    at IDLGen.parseFileContent (/home/awtk/awtk/tools/idl_gen/idl_gen.js:358:14)
    at IDLGen.parseFile (/home/awtk/awtk/tools/idl_gen/idl_gen.js:373:12)
    at IDLGen.parseFolder (/home/awtk/awtk/tools/idl_gen/idl_gen.js:383:12)
    at IDLGen.run (/home/awtk/awtk/tools/idl_gen/idl_gen.js:470:12)
    at /home/awtk/awtk/tools/idl_gen/idl_gen.js:499:11
    at Array.forEach (<anonymous>)
    at Function.gen (/home/awtk/awtk/tools/idl_gen/idl_gen.js:498:9)
    at Object.<anonymous> (/home/awtk/awtk/tools/idl_gen/index.js:24:8)
    at Module._compile (node:internal/modules/cjs/loader:1155:14)`

Fix : check methods before push.
```
--- a/awtk/awtk/tools/idl_gen/idl_gen.js
+++ b/awtk/awtk/tools/idl_gen/idl_gen.js
@@ -163,7 +163,9 @@ class IDLGen {
       let cls = this.getClass(method.className);
       cls.methods.push(method);
     } else if (this.cls) {
-      this.cls.methods.push(method);
+      if(this.cls.methods){
+        this.cls.methods.push(method);
+      }
```

RainMic1019, 2023-02-08:
I failed to reproduce this problem according to your fix (check methods before push).
Platform:  Windows 10.
nodejs:     v16.14.2.

RainMic1019, 2023-02-08:
What environment do you use?

AzAlam1, 2023-02-14:
Platform: Linux
GCC: mips-linux-gnu-gcc (v4.7.2)
nodejs: v8.19.2
target : mips

RainMic1019, 2023-02-14:
The nodejs version you are using is too low. AWTK requires nodejs version>=10.0.0. It is recommended to update nodejs to the latest version and try again.