zlgopen/awtk #816: AGGE相关的死机

Sokyx, 2023-03-18:
1、我这边遇到一个agge死机，应用用的backtrace捕获堆栈，每次死机的堆栈都一样，如下：
 =======System error [Segmentation fault], Stack trace: size = 4==========
[00] dvp_main_t() [0x17270] 
 [01] /lib/libc.so.6(+0x25550) [0x9e898550] 
[02] /system/lib/libawtk.so(_ZN4agge17vector_rasterizer4sortEb+0x1a4) [0xa5f05b4c] 
[03] /system/lib/libawtk.so(_Z12renderStrokeIN4agge12pixel32_bgraEEvPvP8NVGpaint26NVGcompositeOperationStateP10NVGscissorffPK7NVGpathi+0xd0) [0xa5efa444] 
2、每次死机的代码定位也都一样，都是在刚打开窗口或者刚退出窗口的时候（窗口都有平滑动画），执行到window_manager_default_snap_prev_window---widget_paint----widget_paint_impl----widget_on_paint_border函数绘制边框的时候死机。正好平时也有个绘制边框的异常，就是主界面是有system_bar的，第二个界面是全屏的，里面放着三个tab_button子标签页，tab_button都是上边框，边框宽度为8，经常在主界面进入第二个界面的时候，会出现边框一瞬间绘制成对角线的，然后变正常top_border，这个xml如下：
<style name="tab_call" bg_color="#454556" icon_at="center"  spacer="5"  round_radius_bottom_left="20">
    <normal     />
    <pressed    />
    <over       />
    <focused	/>
    <normal_of_active     border="top" border_color="#9facff" border_width="8" bg_color="#4e5067"/>
    <pressed_of_active    border="top" border_color="#9facff" border_width="8" bg_color="#4e5067"/>
    <over_of_active       border="top" border_color="#9facff" border_width="8" bg_color="#4e5067"/>
  </style>

<tab_control x="0" y="0" w="100%" h="100%">
	<tab_button_group x="0" y="0" w="100%" h="100" children_layout="default(r=1,c=0,m=5,s=5)">
             <tab_button name="tab_call" style="tab_call" icon="tab_call" active_icon="tab_call_h"/>
             <tab_button name="tab_unlock" style="tab_unlock" icon="tab_pwd" active_icon="tab_pwd_h"/>
            <tab_button name="tab_addrbook" style="tab_addrbook" icon="tab_addrbook" active_icon="tab_addrbook_h"/>
        </tab_button_group>
       <pages  name="home_pages" x="0" y="100" w="100%" h="-100">
                  ...................
       </pages>
</tab_control>
3、死机的时候见过几次提示：called after throwing an instance of 'std::bad_alloc'   what():  std::bad_alloc abort死机，看样子是C++的vector导致的死机，和第一点的死机堆栈能对应上
4、用过sanitize-address工具检测过，没发现什么问题，不过awtk源码我没用此工具编译，我都是把自己的代码用这个工具编译后运行，因为我认为是我自己的代码影响到了awtk运行
5、试着看agge相关的代码，但是看不太懂，而且没看到哪里有调用到agge相关的代码，请问这个死机问题我应该进一步怎么排查啊，感觉这些打印信息挺明显的，就是不知道什么原因引起的

Sokyx, 2023-03-20:
目前定位的死机：widget_on_paint_border----widget_draw_border----widget_stroke_border_rect----canvas_stroke_rounded_rect_ex----ffr_draw_stroke_rounded_rect_ex----ffr_draw_stroke_rounded_rect----ffr_draw_rounded_rect_draw_stroke_with_vg----vgcanvas_translate(第二次的那个)，最后运行到vgcanvas_translate死机，这个函数我看了下里面就是浮点运算，然后memcpy，但是不知道为啥提示的是aggevector死机

WNsACE, 2023-03-20:
崩溃前，你有没有查看系统里面的内存是否够？系统还有的多少内存？
1，1 楼的问题里面看到的崩溃信息应该 renderStroke 函数导致的，你可以在 renderStroke 函数里面写个 printf ，看看代码走到哪里了吗？
2，2 楼的信息感觉好像是堆溢出了，可以尝试手动调大一下 GUI 线程的堆大小，看看能不能修复。

看完上面的说法，感觉比较像是堆溢出了。

Sokyx, 2023-03-20:
1、内存是够的，可用内存还剩700MB
2、我进一步跟踪一下吧，看堆栈是sort死机，但其实我界面是有圆角和边框的button，但是那些都是border_all边框以及四周都是圆角的button，这三个tab_button用的是top_border和其中一个角是圆角。另外我那三个tab_button不要圆角好像就不会死机，画边框也不会出现那个对角线异常了
3、是要我调main_loop_simple_run函数的堆大小吗？感觉这个应该不会小啊，我用的RK的linux平台，这些都是默认的，我界面都是简单的操作，不至于默认的都不够用吧

WNsACE, 2023-03-20:
1，sort 是哪里的函数？你可以把完整的堆栈信息搞出来看看？
2，如果不是堆栈的问题的话，你看看把 nanovg.c 文件中的 NVG_MAX_STATES 改成 64 看看，会不会出现问题？

Sokyx, 2023-03-21:
1、我看agge的cpp相关代码里面有调用sort函数，但是不知道哪里来的，我那个堆栈是应用调用backtrace捕获得到的，只能打印那么多了
2、具体是不是堆栈问题，不太清楚，后面我继续排查一下，也把你说的这个宏改大一点试试。
3、tab_button的样式设成top_border，去掉左下角或右下角圆角，都用直角的，就没死机了，之前是死机很频繁

WNsACE, 2023-03-21:
1，关于 sort 的话，我感觉可能是 agge 里面的 pod_vector.h 文件中的 grow 函数中的 new 函数失败了？你可以去查一下看看。
2，如果把 NVG_MAX_STATES 改大了，还是不行的话，感觉还是可能是堆栈的问题，因为 vgcanvas_translate 函数里面没啥东西了，不应该在哪里会死掉的。

Sokyx, 2023-03-21:
1、我在grow函数加了打印，发现new的值突然变的很大，然后就死机了，如下打印：
inline void pod_vector<T>::grow(count_t by)
	{
		count_t size = this->size(), new_capacity = capacity();

		new_capacity += agge_max(2 * by > new_capacity ? by : new_capacity / 2, 1u);
printf("1111111111 %s size=%d new_capacity=%d\n", __func__, size, new_capacity);
		T *buffer = new T[new_capacity], *p = buffer;
		printf("2222222222222 %s size=%d new_capacity=%d\n", __func__, size, new_capacity);
		for (iterator i = _begin; i != _end; )
			*p++ = *i++;
		if (_begin != 0) {
			delete []_begin;
		}
		_begin = buffer;
		_end = _begin + size;
		_limit = _begin + new_capacity;
	}

1111111111 grow size=0 new_capacity=1
2222222222222 grow size=0 new_capacity=1
1111111111 grow size=1 new_capacity=2
2222222222222 grow size=1 new_capacity=2
1111111111 grow size=0 new_capacity=1
2222222222222 grow size=0 new_capacity=1
1111111111 grow size=1 new_capacity=2
2222222222222 grow size=1 new_capacity=2
1111111111 grow size=1 new_capacity=20
2222222222222 grow size=1 new_capacity=20
1111111111 grow size=9 new_capacity=476
2222222222222 grow size=9 new_capacity=476
1111111111 grow size=16 new_capacity=3310059
2222222222222 grow size=16 new_capacity=3310059
1111111111 grow size=25617 new_capacity=4965088
2222222222222 grow size=25617 new_capacity=4965088
1111111111 grow size=0 new_capacity=25717
2222222222222 grow size=0 new_capacity=25717
1111111111 grow size=0 new_capacity=103
2222222222222 grow size=0 new_capacity=103
1111111111 grow size=0 new_capacity=1655506
2222222222222 grow size=0 new_capacity=1655506

2、那这样的话，应该就不用再改NVG_MAX_STATES宏了吧，跟这个宏没关系了