zlgopen/awtk #825: Potentially missing null check

szsam, 2023-04-23:
The result value of some functions are often checked for nullness, but not always. Since the value is mostly checked, it is likely that the function can return null values in some cases, and omitting the check could crash the program.
For example, the result of this call to widget_get_window is not checked for null, but 76% of calls to widget_get_window check for null.
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/ext_widgets/combo_box_ex/combo_box_ex.c#L39

Below is the list of potentially missing null checks that I found.
## In non-static functions
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/ext_widgets/image_animation/image_animation.c#L317-L318 (But this one seems okay to fine.)

## In static functions
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/ext_widgets/combo_box_ex/combo_box_ex.c#L39
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/base/hscrollable.c#L58
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/conf_io/conf_ini.c#L181
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/input_methods/input_method_default.inc#L292
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/window_manager/window_manager_default.c#L597
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/src/window_manager/window_manager_default.c#L618

xianjimli, 2023-04-23:
已修改。非常感谢！

szsam, 2023-04-23:
Fixed in 7ff2d2e

szsam, 2023-04-25:
Found one more missing null check @xianjimli 
https://github.com/zlgopen/awtk/blob/f534853b32907572a790cb29a19c09e49ced8d13/tools/common/utils.c#L215

szsam, 2023-04-27:
Fixed in 68afe64498904b5d43b99a925486ce58ff2576e1

closed 2023-04-27 by szsam