zlgopen/awtk #846: 请教个问题？

ytchw101, 2023-07-10:
你好，目前我这边是嵌入式工程移植awtk  + RTT + 文件系统  且从程序SD卡启动（根目录设置为SD卡根目录 tk_init(lcd_w, lcd_h, APP_MOBILE, NULL, ROOT_DIR_SD);）  运行显示一切正常！   字体管理（解析采用 stb方式即打开宏 WITH_STB_FONT）都正常
我想问的是： 我改成freetype方式加载 （不想用 FT_New_Memory_Face 从内存加载 因为从内存加载 需要把字体先预加载到内存感觉 慢）  我使用下面：  
请问： 不管我自己移植一个freetype2源码进来或者使用系统自带的freetype.h ，总是在FT_New_Face 时返回错误码2（即FT_Err_Unknown_File_Format）
FT_Err_Unknown_File_Format: 表示可以打开,读这个文件,但是不支持它的格式,
			   其他非零的返回值,表示不能打开跟读该文件.返回零表示装载成功
我移植了2个版本freetype2 源码最终都是这个错误提示， 不知道是跟系统哪里冲突吗？还是什么原因，不得其解？
freetype2 解析 ttf矢量字体 应该没问题啊，文件路径我也检查应该都是对的，我换成xxx.ttc字体也是不行，不知为什么总提示不支持此格式？

示例：
FT_Library    pFTLib        =  NULL;
FT_Face       pFTFace       =  NULL;
FT_Error      error         =   0 ;
	error  =  FT_Init_FreeType(&pFTLib);
	rt_kprintf("FT_Init_FreeType  error=%d \n",error);
	if(file_exist("/SD/songti.ttf")) //存在
	{
		  rt_kprintf("songti.ttf is exist!\r\n");
	}
	else
	{
		  rt_kprintf("Not found songti.ttf!\r\n");
			FT_Done_FreeType(pFTLib);
			pFTLib  =  NULL;
		  return 0;
	}
	error  =  FT_New_Face(pFTLib, "/SD/songti.ttf", 0, &pFTFace);
      .....

ytchw101, 2023-07-12:
已搞定！ 打扰了........

closed 2023-07-12 by ytchw101

hqulxw123, 2024-10-18:
> 已搞定！ 打扰了........

请问怎么解决的！我也遇到同样的情况

ytchw101, 2024-10-18:
具体我还真记不得了，好像还是代码写的有问题...
注意1.pFTLib 只初始化一次
&nbsp; &nbsp; &nbsp; 2.pFTFace 注意 FT_Done_Face
3.字体路径保证没错

参考：
int LoadFace(const char* fontname)
{
	 FT_Error error = 0;
	 char font_path[100] = {0};
	 rt_sprintf(font_path, "%s%s%s", ROOT_SD_TTF, fontname,".ttf");
			
	 if(0 == strcmp("msyh", fontname))
	 {
		&nbsp; return -1;
	 }
	&nbsp;
&nbsp; &nbsp;if(NULL == pFTFace)
	 {
	&nbsp; &nbsp; error = FT_New_Face(pFTLib, font_path, 0, &amp;pFTFace);
		 	memset(face_fontname, 0, 64);
		&nbsp; memcpy(face_fontname, fontname, strlen(fontname));
//		&nbsp; rt_kprintf("FT_New_Face error=%d,face_fontname：%s\r\n", error,face_fontname);
	 }
	 else
	 {
	&nbsp; &nbsp; &nbsp;if(strcmp(fontname, face_fontname) == 0)
			 {
//				&nbsp; rt_kprintf("Same name not New Face...\r\n");
			&nbsp; &nbsp; return error;
			 }
			 else
			 {
				&nbsp; FT_Done_Face(pFTFace);
					pFTFace&nbsp; =&nbsp; NULL;
				&nbsp;
			&nbsp; &nbsp; error = FT_New_Face(pFTLib, font_path, 0, &amp;pFTFace);&nbsp;
				 	memset(face_fontname, 0, 64);
					memcpy(face_fontname, fontname, strlen(fontname));
//				&nbsp; rt_kprintf("FT_New_Face error=%d,face_fontname：%s\r\n", error, face_fontname);
			 }
	 }
	&nbsp;
	 return error;
}