zlgopen/awtk #847: 滑动scrollview

songmatsu2020, 2023-07-20:
滑动listview或者scrollview时，会改变item的状态为press或者focus

WNsACE, 2023-07-24:
不太明白你的意思，但是我自己实验了一下，并没有你说的情况？你可以否提供一下测试例子？
[image]

songmatsu2020, 2023-07-24:
我这边也是一样的控件布局，滑动的起始位置要在edit控件范围里面就会变成点击事件。同样的，滑动的起始位置所在的item的状态也会改变（会从normal改为pressed），给人一种点击的效果。

WNsACE, 2023-07-24:
你给一个测试例子看看？我上面那个是用最新的 awstudio 来简单做的例子来的，如果鼠标点击到 edit 的话，理论上是拖拽不了的。

songmatsu2020, 2023-07-25:
就和你一样的布局，你的演示例子中，屏幕滑动的起始位置并不是在edit控件范围里，所以可以拖动，我这边也可以正常拖动。但是，屏幕滑动的起始位置是在edit控件范围里的话，就会变成点击edit。你从这个范围里开始滑动就会变成点击edit的效果，怎么避免滑动变点击的情况出现。
[image]

WNsACE, 2023-07-27:
如果点中 edit 控件的话，是无法滑动的啊，

WNsACE, 2023-08-03:
在安卓的情况和 pc 的是一样的阿，我附上测试视频和测试 apk，都是用最新的 awtk 编译出来的。
[app-debug.zip](https://github.com/zlgopen/awtk/files/12246070/app-debug.zip)

https://github.com/zlgopen/awtk/assets/26028060/8b2af30b-607a-42ba-949c-a72b2c824262

WNsACE, 2023-08-11:
上面也说过了，edit 获取焦点后是无法滑动的，gui 的滚动控件基本都是这种逻辑的，在 11s 的时候，我是故意点中 edit 让其获得焦点的，这个时候滚动控件就是不能滚动的，要不然 edit 的位置突变，用户可能就无法输入了，我不知道你在哪里看到这种选中了 edit 还可以滚动的效果，你可以给出例子吗？