zlgopen/awtk #851: 请问问题label_resize_to_content？

ytchw101, 2023-08-10:
先创建一个 widget_t* cont_obj = canvas_widget_create(page_main, 0, 50, 800, 300); 然后
    widget_t* label = label_create(cont_obj, 10, 10, 300, 150); //创建位置(x,y)相对于cont_obj左上角
    // label_resize_to_content(label, 10, 0xFFFF, 10, 0xFFFF);
    widget_set_style_int(label, "font_size", 150);
    widget_set_style_str(label, "text_color", "black");
    widget_set_text(label, L"123");
可以显示，但是使用
 label_resize_to_content(label, 10, 0xFFFF, 10, 0xFFFF);

ytchw101, 2023-08-10:
2. 另外请教下， widget_set_text 文字显示后，字体边缘很平滑，这个平滑处理 是在源码哪里处理的？ 我现在自己font_get_glyph 获得的字摸数据再 组装后 显示很边缘很粗糙。。。

xianjimli, 2023-08-13:
需要指定控件宽度。

ytchw101, 2023-08-14:
> 
哦 好吧！   另外请问  程序使用  WITH_STB_FT 方式 字体解析后 对字 边缘平滑处理 是在哪处理的啊？

xianjimli, 2023-08-16:
没有特殊处理啊