zlgopen/awtk #861: 最新版本按键功能异常

zhaoyangxia1990, 2023-11-21:
更新到最新版本版本 commit b6fae3eb840b7e4241dc2ffd2562dc351b851808
实体按键输入功能不正常，版本回退按则正常。

WNsACE, 2023-11-22:
1，具体哪里不正常，不正常的点是什么？
2，回退到哪一个版本是正常的？

zhaoyangxia1990, 2023-11-22:
1.不正常体现在之前能正常触发按键绑定的命令，现在不能触发，每次按键按下log打印unkown code: e.type=0 code=0 value=0
2.回退到commit 3fabccbfd135c26d47e78c2e5ba36dba71aecd9c正常，中间版本没试过。

WNsACE, 2023-11-22:
你应该是用 linux-fb 的适配层吧，
因为 unkown code: e.type=0 code=0 value=0 应该是 linux-fb 里面打印出来的信息，就是无法识别驱动读取的按键类型而打印出来的，如果是这个打印信息应该和 awtk 的版本没有关系的吧。

zhaoyangxia1990, 2023-11-22:
看了awtk-linux-fb下面的awtk-port/input_thread/input_thread.c的input_dispatch_one_event都是通过标准的read读取输入按键输入设备，输入设备驱动没有修改过，之前都正常的。而且回退AWTK版本可以恢复正常

WNsACE, 2023-11-22:
你的按键是物理按键？还是鼠标/触摸点击？你要么跟一下啊，linux-fb 的代码，看看为啥会触发这个打印？正常来说触发这个打印的话，awtk 应该是收不到事件的，所以 awtk 就没有反应了

WNsACE, 2023-11-24:
我今天使用linux的虚拟机来测试，用的 awtk 和 linux-fb 都是最新的，没有发现什么问题，下面是测试的 gif
[image]

zhaoyangxia1990, 2023-11-24:
不好意思我没说清楚，是物理按键不是触摸键也不是键盘鼠标按键。
从底层读取的按键事件如下，EV_KEY-按键按下 -》EV-SYN-按键同步 -》 EV_KEY-按键抬起
zhaoyangxia, type = 1, code = 60, value = 1
zhaoyangxia, type = 0, code = 0, value = 0
zhaoyangxia, type = 1, 1, code = 60, value = 0

问题出在EV_SYN这个分支
[image]
之前也是一样的按键事件，在EV-SYN这个分支也会打印unkown code: e.type=0 code=0 value=0，但是按键功能还是正常的，所以调用input_dispatch_one_event(run_info_t* info) 时传参应该是跟之前版本有变化

closed 2023-11-25 by zhaoyangxia1990