zlgopen/awtk #864: 页面提示找不到view_model

Pclumsy0540, 2023-12-29:
你好，我这边页面快速切换的时候有概率出现这种提示，找不到对应的view model，可以帮忙看看吗
binding_context_awtk_create_view_model not found view_model history
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket
expr error: expected open bracket

xianjimli, 2024-01-06:
这个好像不是AWTK/AWTK-MVVM打出来的，我搜了一下，没有找到。

Pclumsy0540, 2024-01-08:
binding_context_awtk_create_view_model not found view_model 这一句打印是在awtk-mvvm/src/mvvm/awtk/binding_context_awtk.c里的，然后下面的error是在awtk/src/tkc/expr_eval.c里的，帮忙瞅瞅呢

> 这个好像不是AWTK/AWTK-MVVM打出来的，我搜了一下，没有找到。

xianjimli, 2024-01-08:
还在用 expr_eval.c，那是多少年前的版本啊，你用最新版本试试。如果在PC上重现，发给我看看看。