zlgopen/awtk #871: Missing null check

szsam, 2024-03-03:
The return values of the following functions should be checked against NULL.
https://github.com/zlgopen/awtk/blob/1d8140d3d61126deaa2a36ccefa14cfd4311b0a8/tests/assets_manager_test.cc#L366
https://github.com/zlgopen/awtk/blob/1d8140d3d61126deaa2a36ccefa14cfd4311b0a8/tests/conf_ubjson_test.cc#L82

xianjimli, 2024-03-04:
已修改，谢谢。

closed 2024-03-04 by szsam

Machiry, 2024-03-25:
Hey @xianjimli ,

I am Aravind Machiry, an Assistant Professor at Purdue's ECE Department.

This pull request (along with several others) is the result of our ongoing research work (along with @szsam) to improve the security of open-source embedded projects.

We are doing a short (5-minute) survey to understand the use of static analysis tools in embedded projects.

It would greatly benefit our research if you could fill out this anonymous survey: https://purdue.ca1.qualtrics.com/jfe/form/SV_0OnXfr5plPe1QCa

Thank you,
Aravind

xianjimli, 2024-03-30:
@Machiry OK, I finished the survey.