zlgopen/awtk #880: 关于icon对齐逻辑

Tracker647, 2024-06-25:
1.有一个业务逻辑是单选，单选不选和选中的样式分别如下：
[image]
选中的样式设计，能想到的最简单的实现方法就是设置icon把三角形勾加进来，然后设置右上角对齐，但是我看designer里面只规定了上下左右中五个位置，似乎不支持直接选中。
这样只能绕弯改用代码实现，但这样在mvvm开发模式上又有解耦的问题。
想改成独立图片用xywh布局实现右对齐，但是单选按钮不支持子控件，如果单选按钮和图片封装在同一个view里，封装之间又没法实现单选的逻辑，因为各自隔了一个view
无论哪种绕弯，都十分繁琐。
这个对齐方式是否能加入？

2.感觉icon的对齐逻辑很诡异，比如单选按钮，圆框按理是左对齐，可默认却是auto，如果改成left布局，调节按钮的高度，字体和圆框的距离居然还会随着高度改变（预想效果是只改变显示范围），给界面设计带来很大不便，这属于开发bug吗，可否解决或者有简单设置规避？

[image]
[image]

xianjimli, 2024-06-26:
> icon_at 增加了 left_top/left_bottom/right_top/right_bottom四种位置。

你更新awtk试试，谢谢。designer的问题不归我管，你在群里问吧

Tracker647, 2024-06-26:
好的，谢谢。
响应很快啊，是全天负责维护这个项目吗？

xianjimli, 2024-06-26:
最近两年在做其它项目，不过我的时间比较灵活。

xianjimli, 2024-06-28:
把icon和xml贴上来我试试。

Tracker647, 2024-06-28:
home_page.xml
```xml
<window name="home_page">
  <button name="button" x="228" y="189" w="168" h="62" style:normal:icon_at="right_top" style:normal:icon="icons_Group_3604" text="Button"/>
</window>
```
[image]

xianjimli, 2024-06-28:
改了，你更新awtk试试。谢谢。

Tracker647, 2024-06-29:
好的 布局效果正常了 谢谢

Tracker647, 2024-08-01:
awtk的icon_at属性的更新能不能加到designer？还是有没有什么自己能更新进去的方法？
现在designer一动全局样式就会修改default.xml，把设置了icon_at新属性的样式破坏掉，很烦。