zlgopen/awtk #885: 能不能给窗口新增两个属性move_focus_parent_key和move_focus_child_key？

Tracker647, 2024-07-26:
需求是像这样的菜单页面，它分多个栏，一个栏里面可能有很多控件，对于纯按键控制的设备，一般交互方式就是上下左右切换焦点，然后enter键选中/聚焦子页面内部的焦点，esc键回到父级页面焦点。
[image]
目前看awtk是只有平级控件，基于坐标的焦点切换的，对于基于控件树的父级-子级的焦点切换目前还没有，只能自己实现。
我想原理应该好实现，比如做一个栈，每次按enter键时候把原先窗口控件树所有的focus控件放到数组中并压栈，然后使用下一层focus控件的数组用于切换，按esc就出栈之前的focus数组覆盖当前focus数组。
设想按enter的话目前就聚焦第一个子控件（如果能做到enter的话回到之前聚焦的子控件那更好，不过我暂时没想到实现方法，还有就是，如果这个子控件enter有冲突的话，比如按钮，如何处理？）。
不知这部分后继可有计划？

Tracker647, 2024-07-28:
我今天弄了一版方案，改了awtk，自己目前测了下没问题，你有空看看，我直接把整个awtk库压缩发过去还是能push？

xianjimli, 2024-07-31:
你放个 patch 到这里吧，有需要的自己先打上。

Tracker647, 2024-07-31:
patch：
[tree_focus.patch](https://github.com/user-attachments/files/16444248/tree_focus.patch)
不知道为什么中文输出会乱码：附带一下源文件：
widget_tree_focus.h
```
#ifndef WIDGET_TREE_FOCUS_MANAGER_H
#define WIDGET_TREE_FOCUS_MANAGER_H
#include "tkc/darray.h"
#include "base/widget.h"
BEGIN_C_DECLS
/**
 * @brief 当窗口基于控件树的"层级逻辑"进行焦点切换时，采用此类进行管理
 * （如果切换到子控件，focusable控件将切换为子控件内部的所有的focusable控件，并屏蔽外部父层及以上的focusable控件）
 * 在xml上的使用方法例：
 * <window name="focus_test_window" style:normal:bg_color="#1A2F4A" move_focus_up_key="up" move_focus_down_key="down" move_focus_left_key="left" move_focus_right_key="right" move_focus_as_tree_enable="true" move_focus_children_key="RETURN" move_focus_parent_key="ESCAPE">
 ...
 </window>
 * 若想根据控件树层级进行焦点切换，move_focus_as_tree_enable需在xml指明并设为true
 * 
 */
typedef struct {
    widget_t *win;
    /**
     * @brief 进入子控件层时，将上层开始的所有focusable控件list压栈到此
     * 
     */
    darray_t *g_focus_widget_list_stack;
    /**
     * @brief 进入子控件层时，将前一个聚焦的控件的压栈至此
     * 
     */
    darray_t *g_focus_widget_stack;
} widget_tree_focus_manager_t;

widget_tree_focus_manager_t* widget_tree_focus_manager_create(widget_t *win);
ret_t widget_tree_focus_move_parent(widget_tree_focus_manager_t* manager, widget_t *focused_widget);
ret_t widget_tree_focus_move_children(widget_tree_focus_manager_t* manager, widget_t *focused_widget);
ret_t widget_tree_focus_manager_destroy(widget_tree_focus_manager_t *manager);

END_C_DECLS
#endif  /* WIDGET_TREE_FOCUS_MANAGER_H */
```

widget_tree_focus.c
```
#include "widget_tree_focus_manager.h"
#include "tkc/mem.h"
#include <stdbool.h>

static ret_t widget_print_all_focusable_widgets(darray_t *all_focusable)
{
	return_value_if_fail(all_focusable, RET_FAIL);
	for (int i = 0; i < all_focusable->size; i++)
	{
		if (all_focusable->elms[i])
		{
			printf("widget_print_all_focusable_widgets>find widget name: %s at %#x\r\n", WIDGET(all_focusable->elms[i])->name, WIDGET(all_focusable->elms[i]));
		}
	}
	return RET_OK;
}

static ret_t widget_get_all_focusable_widgets(widget_t *widget, darray_t *all_focusable)
{
	widget_foreach(widget, widget_on_visit_focusable, all_focusable);
	return_value_if_fail(widget != NULL, RET_FAIL);
	return_value_if_fail(all_focusable != NULL && all_focusable->size > 0, RET_FAIL);
	return RET_OK;
}

static ret_t widget_set_all_focusable(darray_t *all_focusable, bool flag)
{
	return_value_if_fail(all_focusable, RET_FAIL);
	for (int i = 0; i < all_focusable->size; i++)
	{
		if (all_focusable->elms[i])
		{
			widget_set_focusable(WIDGET(all_focusable->elms[i]), flag);
		}
	}
	return RET_OK;
}

static ret_t widget_set_all_focused(darray_t *all_focusable, bool flag)
{
	return_value_if_fail(all_focusable, RET_FAIL);
	for (int i = 0; i < all_focusable->size; i++)
	{
		if (all_focusable->elms[i])
		{
			widget_set_prop_bool(WIDGET(all_focusable->elms[i]), WIDGET_PROP_FOCUSED, flag);
		}
	}
	return RET_OK;
}

widget_tree_focus_manager_t* widget_tree_focus_manager_create(widget_t *win){
    return_value_if_fail(win != NULL, NULL);
    widget_tree_focus_manager_t *manager = TKMEM_ZALLOC(widget_tree_focus_manager_t);
    manager->win = win;
    widget_ref(win);
    manager->g_focus_widget_list_stack = darray_create(10, NULL, NULL);
    manager->g_focus_widget_stack = darray_create(10, NULL, NULL);
    return manager;
}

ret_t widget_tree_focus_move_parent(widget_tree_focus_manager_t* manager, widget_t *focused_widget){
    return_value_if_fail(manager != NULL, RET_BAD_PARAMS);
    darray_t *g_focus_widget_list_stack = manager->g_focus_widget_list_stack;
    darray_t *g_focus_widget_stack = manager->g_focus_widget_stack;
    widget_t *win = manager->win;
    ret_t ret = RET_OK;
    if (g_focus_widget_list_stack->size == 0)
    {
        printf("no focusable parent widget\r\n");
        return RET_STOP;
    }
    darray_t *parent_focusable_widget_list = darray_pop(g_focus_widget_list_stack);
    focused_widget = darray_pop(g_focus_widget_stack);
    ret = widget_set_all_focusable(parent_focusable_widget_list, TRUE);
    ret = widget_set_all_focused(parent_focusable_widget_list, FALSE);
    ret = widget_set_focused(focused_widget, TRUE);

    ret = darray_destroy(parent_focusable_widget_list);
    return ret;
}

ret_t widget_tree_focus_move_children(widget_tree_focus_manager_t* manager, widget_t *focused_widget){
    return_value_if_fail(manager != NULL, RET_BAD_PARAMS);
    darray_t *g_focus_widget_list_stack = manager->g_focus_widget_list_stack;
    darray_t *g_focus_widget_stack = manager->g_focus_widget_stack;
    widget_t *win = manager->win;
    ret_t ret = RET_OK;
    darray_t *parent_focusable_widget_list = darray_create(10, NULL, NULL);
    widget_get_all_focusable_widgets(win, parent_focusable_widget_list);
    darray_t *child_focusable_widget_list = darray_create(10, NULL, NULL);
    widget_get_all_focusable_widgets(focused_widget, child_focusable_widget_list);
    darray_remove(child_focusable_widget_list, focused_widget);

    if (child_focusable_widget_list->size > 0)
    {
        widget_set_all_focusable(parent_focusable_widget_list, FALSE);
        widget_set_all_focusable(child_focusable_widget_list, TRUE);
        widget_set_all_focused(parent_focusable_widget_list, FALSE);
        widget_set_focused(WIDGET(child_focusable_widget_list->elms[0]), TRUE);

        darray_push(g_focus_widget_list_stack, parent_focusable_widget_list);
        darray_push(g_focus_widget_stack, focused_widget);
    }
    else
    {
        printf("no focusable child widget\r\n");
    }
    ret = darray_destroy(child_focusable_widget_list);
    return ret;
}

ret_t widget_tree_focus_manager_destroy(widget_tree_focus_manager_t *manager)
{
    return_value_if_fail(manager != NULL, RET_BAD_PARAMS);
    darray_destroy(manager->g_focus_widget_stack);
    darray_destroy(manager->g_focus_widget_list_stack);
    widget_unref(manager->win);
    TKMEM_FREE(manager);
    return RET_OK;
}
```

Tracker647, 2024-07-31:
测试程序：
[release.zip](https://github.com/user-attachments/files/16444430/release.zip)

Tracker647, 2024-08-09:
BTW：如果我修改了一个历史评论，邮箱上会重新推送这个评论吗？edit一次就推送一次？

Tracker647, 2024-08-10:
更新了一下patch， 现在可以给没父级焦点和子级焦点的情况设置回调，更新这个的原因是业务需要在没有父级焦点可退的时候需要navigate_back。
发现只要在git bash去执行diff指令生成的补丁就能打了，估计是终端编码问题。
` git diff cf5e93980e0d86ef456725f820540114ecc87876..3489cfc1781fda44b6ba32ee278f68bf408c7d45 > treefocus.patch`
[treefocus.patch](https://github.com/user-attachments/files/16569796/treefocus.patch)