zlgopen/awtk #886: text_selector无法实时滚动问题

Nice2Link, 2024-07-28:
大家好，我使用的AWTK版本是1.6.0，使用text_selector控件做时间日期设置界面的时候，有些问题，希望大家指导下，谢谢！
问题如下：
（1）触屏上下点击“Year”控件的时候，不会马上刷新显示加减后的值，需要点击另外的一个“month”控件之后，才会刷新上一个“Year”控件，其他的text_selector控件也是这样。
（2）在“Year”控件上做拖动(MOVE)的时候，也不会实时刷新显示值变化移动的过程，最后也是需要点击另外一个“month”控件之后才会显示刷新“Year”控件拖动到最后停留的值。

- UI的xml文件如下：
`  <window name="time_setting" style:normal:bg_color="#3F3F3F" style:normal:bg_image="">
  <text_selector name="year" x="82" y="214" w="113" h="251" style:normal:border_width="2" style:normal:font_size="26" style:normal:bg_color="#3A3838" feedback="false" focusable="false" loop_options="true" style:normal:text_color="#FFFFFF" style:normal:fg_color="#757070" style:normal:mask_color="#BFBFBF" style:normal:border_color="#FFFFFF" options="1970-2099" selected_index="54"/>  
  <text_selector name="month" x="252" y="214" w="113" h="251" style:normal:border_width="2" style:normal:font_size="26" loop_options="true" style:normal:text_color="#FFFFFF" style:normal:bg_color="#3A3838" style:normal:mask_color="#BFBFBF" style:normal:border_color="#FFFFFF" style:normal:fg_color="#757070" options="1-12" selected_index="0"/>  
  <text_selector name="day" x="425" y="214" w="113" h="251" style:normal:border_width="2" style:normal:font_size="26" loop_options="true" style:normal:bg_color="#3A3838" style:normal:border_color="#FFFFFF" style:normal:fg_color="#757070" style:normal:mask_color="#BFBFBF" style:normal:text_color="#FFFFFF" options="1-31" selected_index="0"/>  
  <text_selector name="hour" x="654" y="214" w="113" h="251" style:normal:border_width="2" style:normal:font_size="26" loop_options="true" style:normal:fg_color="#757070" style:normal:bg_color="#3A3838" style:normal:text_color="#FFFFFF" style:normal:mask_color="#BFBFBF" style:normal:border_color="#FFFFFF" options="0-23" selected_index="0"/>  
  <text_selector name="minute" x="826" y="214" w="113" h="251" style:normal:border_width="2" style:normal:font_size="26" loop_options="true" style:normal:mask_color="#BFBFBF" style:normal:fg_color="#757070" style:normal:bg_color="#3A3838" style:normal:text_color="#FFFFFF" style:normal:border_color="#FFFFFF" options="0-59" selected_index="0"/>  
  <label name="year_txt" x="108" y="137" w="55" h="47" style:normal:text_color="#FFFFFF" style:normal:font_size="40" text="年"/>
  <label name="time_line" x="82" y="184" w="860" h="1" style:normal:border_width="2" style:normal:border_color="#FFFFFF" style:normal:border="bottom" text=""/>  
  <label name="month_txt" x="281" y="137" w="55" h="47" style:normal:text_color="#FFFFFF" style:normal:font_size="40" text="月"/>  
  <label name="day_txt" x="453" y="137" w="55" h="47" style:normal:text_color="#FFFFFF" style:normal:font_size="40" text="日"/>  
  <label name="hour_txt" x="683" y="137" w="55" h="47" style:normal:text_color="#FFFFFF" style:normal:font_size="40" text="时"/>  
  <label name="minute_txt" x="855" y="137" w="55" h="47" style:normal:text_color="#FFFFFF" style:normal:font_size="40" text="分"/>  
  <label name="menu_line" x="0" y="50" w="1024" h="3" style:normal:border_width="3" style:normal:border_color="#FFFFFF" style:normal:border="bottom" style:normal:bg_color="#FFFFFF" text=""/>  
  <button name="back_button" x="30" y="5" w="40" h="40" style:normal:border_width="0" style:normal:bg_image="return_48" style:normal:bg_image_draw_type="center" style:normal:bg_color="#f4f4f400" text=""/>  
  <digit_clock name="digit_clock" x="512" y="5" w="250" h="40" format="Y-MM-DD hh:mm:ss" style:normal:border_width="0" style:normal:text_color="#FFFFFF" style:normal:font_name="default" style:normal:font_size="40"/>  
  <label name="menu_txt" x="110" y="0" w="120" h="50" style:normal:text_color="#FFFFFF" style:normal:font_size="45" style:normal:text_align_h="left" text="时间设置"/>  
  <image name="sd_card" x="974" y="5" w="40" h="40" draw_type="center" image="sd_32" style:normal:border_width="0" style:normal:margin_bottom="0"/>  
</window>  `    

- UI界面初始化代码如下：

static ret_t on_back_button_click(void* ctx, event_t* e)   
{
    return ui_winmng_startwin(UI_HOME_PAGE, true);  
}  

static ret_t init_widget(void* ctx, const void* iter) {  
    widget_t* win = WIDGET(ctx);  
    widget_t* widget = WIDGET(iter);  
    const char* name = widget->name;  

    // 初始化指定名称的控件（设置属性或注册事件），请保证控件名称在窗口上唯一
    if (name != NULL && *name != '\0') {
        /* 在此处对指定名称的控件进行注册事件，此处的事件为指针按下 */
        if (tk_str_eq(name, "back_button")) {
            widget_on(widget, EVT_CLICK, on_back_button_click, win);
        } 

    } else {
        CVI_LOGE("widget->name is NULL! name: %s", widget->name);
    }

  return RET_OK;
}

static void init_children_widget(widget_t* widget) {
    widget_foreach(widget, init_widget, widget);
}

ret_t ui_system_setting_open(void* ctx, event_t* e)
{
    widget_t* win = WIDGET(ctx);
    if (win) {
        init_children_widget(win);
    }
    return RET_OK;
}

ret_t ui_system_setting_close(void* ctx, event_t* e)
{
    (void)e;
    widget_t* win = WIDGET(ctx);
    if (win) {

    }
    return RET_OK;
}

xianjimli, 2024-07-28:
最好用最新版本。先在PC上看看能不能重现，如果能重现，发给我看看。

Nice2Link, 2024-07-28:
> 最好用最新版本。先在PC上看看能不能重现，如果能重现，发给我看看。

这个SOC厂家适配过的只有这个1.6.0版本，这些text_selector控件需要做相关的初始化动作吗？我加了widget_invalidate去刷新控件也不行，如下代码也不行：
```
static ret_t on_text_selector_value_changed(void* ctx, event_t* e) {
  widget_t* widget = WIDGET(e->target);
  
  // 获取选中的值
  const char* selected_value = text_selector_get_text(widget);
  log_debug("Selected value: %s\n", selected_value);

  widget_invalidate(widget, NULL);

  return RET_OK;
}

widget_on(widget, EVT_VALUE_CHANGED, on_text_selector_value_changed, win);

```

Nice2Link, 2024-07-28:
> 最好用最新版本。先在PC上看看能不能重现，如果能重现，发给我看看。

我试了，写一个只有text_selector控件的简单demo，这个demo程序在PC上面运行时没有问题的用鼠标拖动操作text_selextor控件都没有问题，，但是在嵌入式linux设备上运行的时候就是有上述问题。请问还有什么思路排查下吗？

谢谢！