zlgopen/awtk #888: mvvm框架项目自定义控件调用style_get_xxx函数会crash

Tracker647, 2024-08-02:
自定义控件手动用scons编译就不会，但是用designer加载之后然后重编项目就会报错，我自定义控件和项目都是用的github上的awtk来编译，非designer内置，不知道awtk designer内部加载是不是写死了。

repo控件：
[awtk-widget-custom-widget.zip](https://github.com/user-attachments/files/16465332/awtk-widget-custom-widget.zip)

repo方法：
1.designer上新建项目并加载custom_widget控件，确保项目和控件使用的都是github的awtk
2.项目编译脚本run.ps1
```
python ./scripts/update_res.py all
scons 
bin/demo.exe
```
3.运行run.ps1，窗口在启动一段时候自动关闭，日志显示`binding_context_clear_bindings_of_widget:338 condition(ctx != NULL && ctx->vt != NULL && widget != NULL) failed!`

Tracker647, 2024-08-02:
怀疑是不是自定义控件和项目用了不同的sdk加载，因为之前发现自定义控件默认用的designer内置sdk, 然后项目用github的awtk sdk, 结果编译同样报binding_context_clear_bindings_of_widget。之前怀疑designer就是用的内置sdk加载，然后改了自定义控件的project.json的awtk路径然后在designer加载复现，也是报binding_context_clear_bindings_of_widget，不知道是不是写死了。

xianjimli, 2024-08-03:
designer的问题，你在群里问吧。designer组的同事一般不会看github。

closed 2024-09-17 by Tracker647