zlgopen/awtk #895: ubuntu怎么链接带版本号的三方库？

Tracker647, 2024-09-07:
在ubuntu20.04上编译项目，项目需要用到poco库。

步骤一：下载Poco库（我用版本为1.9.4），根据其README，./configure然后make, make install, 得到编译的二进制库：
```
./configure --prefix=/my/path/3rdparty/ubuntu/poco-1.9.4 \
      --shared --static --no-tests --no-samples --no-ipv6 \
      --include-path=/my/path/3rdparty/ubuntu/openssl-1.1.1w/include/ \
      --library-path=/my/path/3rdparty/ubuntu/openssl-1.1.1w/lib \
      --omit=Data/ODBC,Data/MySQL
make
make install
```

步骤二：
在项目根据按照[如何引用第三方库](https://gitee.com/zlgopen/awtk/blob/master/docs/how_to_use_3rd_libs.md)的教程编写SConstruct，把用到的库的lib都改成了bin文件名，因为so库都是在lib下
```
DEPENDS_LIBS = CUSTOM_WIDGET_LIBS + [
  {
        'root' : '3rd/poco-1.9.4/',
        'shared_libs' : ['PocoFoundation', 'PocoUtil', 'PocoXML', 'PocoJSON', 'PocoCrypto', 'PocoNet', 'PocoNetSSL']
    }
]

ARGUMENTS['WITH_MVVM'] = 'true'
helper = app.Helper(ARGUMENTS)

APP_SRC = os.path.normpath(os.path.join(os.getcwd(), 'src'))
THIRD_PARTY_SRC = os.path.normpath(os.path.join(os.getcwd(), '3rd'))
APP_CPPPATH = [
  os.path.join(APP_SRC, 'common'),
  os.path.join(APP_SRC, 'view_models'),
  os.path.join(APP_SRC, 'pages'),
  os.path.join(THIRD_PARTY_SRC, 'poco-1.9.4/include')
]

CXXFLAGS = '-std=c++11'

helper.set_deps(DEPENDS_LIBS).add_cpppath(APP_CPPPATH).add_cxxflags(CXXFLAGS).call(DefaultEnvironment)

SConscriptFiles = ['src/SConscript', 'tests/SConscript']
helper.SConscript(SConscriptFiles)
```

步骤三，scons编译，编译成功，但是bin/demo显示`error while loading shared libraries: libPocoFoundation.so.64: cannot open shared object file: No such file or directory`, 看来是不接受软链接，但是Sconstruct脚本里好像固定写死了只能取libxxx.so的xxx了，只能手动把用到的库CV到bin, 不知道有后缀名的情况有没有更好的，自动化的处理办法？

xianjimli, 2024-09-08:
创建一个符号链接。libxxx.so --> libxxx.so.xx

Tracker647, 2025-01-24:
找到两个解决方法：
1.自己写拷贝函数，指定文件夹把库拷过去：
```
import os
import shutil
import subprocess
import sys
import glob
import platform

def copy_binary_libs_to_build_dir(dst_path, lib_list, src_path):
    """
    将 dst_path 中匹配 lib_list 的文件拷贝到 src_path 文件夹。
    对于 Linux 系统，会查找 lib_list 中对应元素的 .so 文件是否存在，
    如果不存在就调用 dst_path 中的 addsymlink.sh 更新软链接。
    Linux 系统情况下，将拷贝对应元素所有的匹配 .so* 的文件（包括软链接和带版本号后缀的 so 文件）。
    """
    PLATFORM = platform.system()
    # 确保目标目录存在
    os.makedirs(src_path, exist_ok=True)

    for lib_name in lib_list:
        # 根据平台确定库文件的后缀
        if PLATFORM == 'Windows':
            lib_file = f'{lib_name}.dll'
            src_lib_path = os.path.join(dst_path, lib_file)
            dest_lib_path = os.path.join(src_path, lib_file)

            # 检查库文件是否存在
            if not os.path.exists(src_lib_path):
                print(f"Warning: Library {lib_file} not found in {dst_path}.")
                continue

            # 拷贝文件
            shutil.copy2(src_lib_path, dest_lib_path)
            print(f"Copied {src_lib_path} to {dest_lib_path}.")

        elif PLATFORM == 'Linux':
            # Linux 系统下，查找 lib_name 对应的 .so 文件
            lib_file_pattern = f'lib{lib_name}.so*'
            src_lib_pattern = os.path.join(dst_path, lib_file_pattern)
            matching_files = glob.glob(src_lib_pattern)

            if not matching_files:
                # 如果找不到 .so 文件，尝试调用 addsymlink.sh 更新软链接
                addsymlink_script = os.path.join(dst_path, 'addsymlink.sh')
                if os.path.exists(addsymlink_script):
                    print(f"Library {lib_file_pattern} not found. Running addsymlink.sh to update symlinks...")
                    try:
                        subprocess.check_call([addsymlink_script], cwd=dst_path)
                    except subprocess.CalledProcessError as e:
                        print(f"Error: Failed to run addsymlink.sh: {e}")
                        sys.exit(1)

                    # 再次查找 .so 文件
                    matching_files = glob.glob(src_lib_pattern)
                    if not matching_files:
                        print(f"Error: Library {lib_file_pattern} still not found after running addsymlink.sh.")
                        sys.exit(1)
                else:
                    print(f"Error: Library {lib_file_pattern} not found and addsymlink.sh does not exist.")
                    sys.exit(1)

            # 拷贝所有匹配的 .so* 文件
            for src_lib_path in matching_files:
                dest_lib_path = os.path.join(src_path, os.path.basename(src_lib_path))
                shutil.copy2(src_lib_path, dest_lib_path)
                print(f"Copied {src_lib_path} to {dest_lib_path}")

        else:
            print(f"Unsupported platform: {PLATFORM}")
            sys.exit(1)
```

2.看了下prepare_depends_libs函数有这么一段，可以指定把对应文件夹给拷贝到BUILD_DIR下：
 if 'needed_files' in lib:
            if Script.GetOption('clean'):
                clear_needed_files(helper, lib['root'], lib['needed_files'])
            else:
                copy_needed_files(helper, lib['root'], lib['needed_files'])

那么SConstruct可以改成这样：
```
DEPENDS_LIBS += [
    {
        'root': BINARY_LIB_PATHS,
        'shared_libs':[],
        'needed_files': ['bin']
    }

]  
app.prepare_depends_libs(ARGUMENTS, helper, DEPENDS_LIBS)
```
效果一致，后者好处是不用写新函数（虽然这种逻辑AI编写成本挺低）

上面两个方法适用于三方库都在同一个文件夹的情况。