zlgopen/awtk #896: AWTK中有没有办法自定义字体式的icon？或者用矢量图自定义图形？

Tracker647, 2024-09-12:
[image]
要实现这么一个效果，定义几个窗口样式，选中这些窗口样式的时候窗口会变成蓝色。
如果是在html中，记得有个叫iconfont的机制，可以把icon当成字体一样使用，如果遇到事件变色的话icon会像字体一样变色，跟控件同步，不需要额外写页面代码，不知道awtk中有没有类似的机制。

现在的做法就是通过[特殊字符](https://cn.piliapp.com/symbol/)网站，导入想要的字符进去，比如写一个这样的选择箱，里面的三角形按钮要求变色，我就直接用了这网站的▶◀两个字符，但这个的方法的缺陷就是，网站的字符有限，部分字符在awtk上还无法显示，设置打出来显示⛝
[image]

问下除了自己专门一个c文件做这部分页面逻辑还有什么好办法吗？mvvm上，实在不想为此多一个c文件。

Tracker647, 2024-09-12:
add：三角形按钮在浏览器的显示和在awtk上不一样，awtk应该和特殊字符网站的一致。
[image]

xianjimli, 2024-09-17:
看看这个是不是你需要的：https://github.com/zlgopen/awtk-widget-shape