zlgopen/awtk #900: mutable_image在ubuntu上cpu占用配置极高

Tracker647, 2024-09-17:
环境：ubuntu20虚拟机， cpu配置：
[image]

一个显示谷歌地图的项目，要移植到awtk，原版QT项目用QImage实现，从网络获取图片然后拷贝到QImage显示，启动成功后最多只占约3%的CPU。
我移植到awtk，思路类似，QImage对标mutable_image, top发现cpu占用居然快到100%。
[image]

位图像素显示格式都是32位。

起初我以为是我用户代码的问题，但是删了自己代码仍旧高cpu占用，后来装了gperftools排查了下，发现瓶颈都集中在mutable_image_prepare绘制相关的段上，如果有做image_copy相关的操作，cpu占用会更高。
然后我把xml上所有mutable_image控件注释掉重新编译，发现cpu占用就回到个位数了，推测cpu的高占用和mutable_image关系很大。

我做了一个demo，把其他逻辑都剥离掉，只建立mutable_image并设置回调，发现cpu占用也有高达40-60%，不管用agge渲染还是gl渲染都差不多。
[image]

我不清楚这是不是框架控件本身的问题，如果是，有何设置或者解决方案？

测试用的demo如下，里面有gperftool的测试数据，测试环境需安装gperftools。
[mutable_image_cpu_test.zip](https://github.com/user-attachments/files/17026393/mutable_image_cpu_test.zip)

还没在嵌入式linux机器上试过，估计表现会很够呛。

xianjimli, 2024-09-17:
图片格式和lcd保持一致，直接用memcpy显示，否则性能上不来。

Tracker647, 2024-10-05:
> 图片格式和lcd保持一致，直接用memcpy显示，否则性能上不来。

图片格式和lcd保持一致
什么意思？project.json的lcd设置不影响程序啊，只能设置bitmap的格式，格式不一致都不可能编译通过，你说的最终机器的lcd吗？要看固定的硬件参数？

直接用memcpy显示
是指不用image_copy直接buffer_lock_for_write对二维数组操作吗？

xianjimli, 2024-10-06:
mutable_image_prepare_image里解码的图片与LCD格式保持一致。

Tracker647, 2024-10-06:
project.json的lcd参数？
```
   "lcd": {
          "width": "800",
          "height": "480",
          "colorDepth": "32bit",
          "colorFormat": "BGR(A)"
        }
```
还是硬件意义上的lcd?
> mutable_image_prepare_image里解码的图片与LCD格式保持一致。

xianjimli, 2024-10-06:
硬件意义上的lcd。framebuffer的格式。

Tracker647, 2024-10-09:
> 硬件意义上的lcd。framebuffer的格式。

这个格式怎么找？是固定不变的吗？

Tracker647, 2024-10-13:
看了下mutable_image源码，跟lcd格式应该没什么关系，有一个16ms的刷新widget的定时器，我用自定义控件加这个定时器的逻辑上去，在ubuntu测试，cpu立马飙到100，估计cpu占用高，定时器频繁刷新占很大一部分比例。
改成了每100ms刷新就好很多了，ubuntu上测试cpu占用降到60%，机器占用从原80%降到70%，好在需求不需要这么高的刷新率。

如果进一步优化，awtk_config_sample.h的HAS_FASTER_MEMCPY之类的宏可以打开。

closed 2024-10-27 by Tracker647

Sokyx, 2025-06-18:
我也是在用到mutable_image的时候CPU占用率很高，导致某些情况下使用起来会造成影响。
prepare如下：
static ret_t uiMediaPlayerListGetVideoFrame(void* handle, bitmap_t* image) {
	mediaPlayerList_t* mediaPlayList = (mediaPlayerList_t*)handle;
	if (mediaPlayList != NULL && mediaPlayList->playState == MEDIA_STATE_PLAYING && mediaPlayList->image != NULL) {
		rect_t r = rect_init(0, 0, image->w, image->h);
		return image_copy(image, mediaPlayList->image, &r, 0, 0);
	}
	return RET_FAIL;
}
我直接调用的image_copy，mediaPlayList->image里面保存的是bitmap_format_t bitmapFormat = lcd_get_desired_bitmap_format(widget_get_canvas(window_manager())->lcd);格式的数据，也就是跟LCD保持一直的吧？
但是CPU占用率很高，4核1.5GHZ，能占用30%CPU太高了。有什么办法能解决吗