zlgopen/awtk #905: msvc环境下，awtk程序使用poco lib崩溃

Tracker647, 2024-10-04:
步骤一：
用msvc编译awtk
步骤二：
designer新建一个awtk项目
步骤三：
安装vcpkg，安装完毕后设置环境变量，然后命令行输入如下：
vcpkg install poco[crypto,netssl]
vcpkg install expat
vcpkg install pcre2
vcpkg install zlib

步骤四：将下好的包从<vcpkg安装目录>/packages 复制粘贴到项目3rd，共有如下包：
```
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----         2024/10/4     11:00                expat
d-----         2024/10/4     10:36                openssl
d-----         2024/10/4     11:00                pcre2
d-----         2024/10/4     10:36                poco
d-----         2024/10/4     11:11                ubuntu
d-----         2024/10/4     11:00                zlib
```

Tracker647, 2024-10-04:
步骤五编写Sconstruct引入：
```
import os
import scripts.app_helper as app
import platform

PLATFORM = platform.system()

CUSTOM_WIDGET_LIBS = []

DEPENDS_LIBS = CUSTOM_WIDGET_LIBS + []

SOURCE_3RD_DEPS_PATH = os.path.normpath(os.path.join(os.getcwd(), '3rd'))

if PLATFORM == 'Linux':
    SOURCE_3RD_DEPS_PATH = os.path.join(SOURCE_3RD_DEPS_PATH, 'ubuntu')
elif PLATFORM == 'Windows':
    SOURCE_3RD_DEPS_PATH = os.path.join(SOURCE_3RD_DEPS_PATH, 'win_msvc')

if PLATFORM == 'Linux':
    DEPENDS_LIBS += [
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'zlib'),
            'shared_libs' : ['z']
        },
        {
            'root': os.path.join(SOURCE_3RD_DEPS_PATH, 'openssl'),
            'shared_libs': ['crypto', 'ssl']
        },
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'poco'),
            'shared_libs' : ['PocoFoundation', 'PocoUtil', 'PocoXML', 'PocoJSON', 'PocoCrypto', 'PocoNet', 'PocoNetSSL']
        }
    ]
elif PLATFORM == 'Windows':
    DEPENDS_LIBS += [
        {
            'root': os.path.join(SOURCE_3RD_DEPS_PATH, 'openssl'),
            'shared_libs': ['libcrypto-3-x64', 'libssl-3-x64']
        },
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'poco/debug'),
            'shared_libs' : ['PocoFoundationd', 'PocoUtild', 'PocoXMLd', 'PocoJSONd', 'PocoCryptod', 'PocoNetd', 'PocoNetSSLd']
        },
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'pcre2/debug'),
            'shared_libs' : ['pcre2-8d']
        },
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'zlib/debug'),
            'shared_libs' : ['zlibd1']
        },
        {
            'root' : os.path.join(SOURCE_3RD_DEPS_PATH, 'expat/debug'),
            'shared_libs' : ['libexpatd']
        }
    ]

APP_CPPPATH = [
    os.path.join(SOURCE_3RD_DEPS_PATH, 'openssl/include'),
    os.path.join(SOURCE_3RD_DEPS_PATH, 'poco/include')
]

helper = app.Helper(ARGUMENTS)
helper.set_deps(DEPENDS_LIBS).add_cpppath(APP_CPPPATH)

app.prepare_depends_libs(ARGUMENTS, helper, DEPENDS_LIBS)
helper.call(DefaultEnvironment)

SConscriptFiles = ['src/SConscript', 'tests/SConscript']
helper.SConscript(SConscriptFiles)

```

步骤六，编译运行，报错：
[image]

Tracker647, 2024-10-04:
用的库在release这里, 和bin一起：
[release.zip](https://github.com/user-attachments/files/17255505/release.zip)

就是在启动页面的init函数上测试了点Poco的api就崩溃：
```
#include"download.h"
#include "Poco/Net/HTTPSClientSession.h"
#include "Poco/Net/HTTPRequest.h"
#include "Poco/Net/HTTPResponse.h"

#include "Poco/Net/SSLManager.h"
#include "Poco/Net/AcceptCertificateHandler.h"

#include "Poco/StreamCopier.h"
#include "Poco/File.h"

#include "Poco/URI.h"
#include "Poco/Exception.h"

void download()
{
    std::string strURI("https://mt1.google.com/vt/lyrs=s&x=208725&y=123184&z=18");
    Poco::URI uri(strURI);

    // std::string path(uri.getPathAndQuery());
    // if (path.empty()) path = "/";

    // Poco::Net::HTTPSClientSession session(uri.getHost(), uri.getPort());
    // printf("Downloading %s(%s:%d)\n", strURI.c_str(), uri.getHost().c_str(), uri.getPort());
}
```

我感觉不是库或者msvc编译器的问题，因为我在Qt和VS上都跑的通，不知道awtk是不是不兼容C++。

Tracker647, 2024-10-04:
环境：
Windows 11
MSVC 1941(Visual Studio 2022)

Tracker647, 2024-10-04:
reproduce在这里：https://github.com/Tracker647/awtk-crash-problem-record/tree/master

xianjimli, 2024-10-07:
看了一下， 我也不知道怎么回事。找个windows高手咨询一下。

Tracker647, 2024-10-07:
我试了很多方法，源码编译64位，32位，乃至直接把cpp文件放到scons去构建都没用，scons里构建源码发现凡是Poco的文件都会报警告，Qt上就正常，搞不好漏了什么编译参数？
```
ArchiveStrategy.cpp
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(86): warning C4273: “Poco::ArchiveStrategy::ArchiveStrategy”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(43): note: 参见“Poco::ArchiveStrategy::ArchiveStrategy”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(93): warning C4273: “Poco::ArchiveStrategy::~ArchiveStrategy”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(44): note: 参见“Poco::ArchiveStrategy::~ArchiveStrategy”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(99): warning C4273: “Poco::ArchiveStrategy::compress”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(54): note: 参见“Poco::ArchiveStrategy::compress”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(105): warning C4273: “Poco::ArchiveStrategy::moveFile”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(58): note: 参见“Poco::ArchiveStrategy::moveFile”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(131): warning C4273: “Poco::ArchiveStrategy::exists”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(59): note: 参见“Poco::ArchiveStrategy::exists”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(154): warning C4273: “Poco::ArchiveByNumberStrategy::ArchiveByNumberStrategy”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(76): note: 参见“Poco::ArchiveByNumberStrategy::ArchiveByNumberStrategy”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(159): warning C4273: “Poco::ArchiveByNumberStrategy::~ArchiveByNumberStrategy”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(77): note: 参见“Poco::ArchiveByNumberStrategy::~ArchiveByNumberStrategy”的前一个定 义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(164): warning C4273: “Poco::ArchiveByNumberStrategy::open”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(79): note: 参见“Poco::ArchiveByNumberStrategy::open”的前一个定义
3rd\win_msvc\poco\src\ArchiveStrategy.cpp(170): warning C4273: “Poco::ArchiveByNumberStrategy::archive”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/ArchiveStrategy.h(80): note: 参见“Poco::ArchiveByNumberStrategy::archive”的前一个定义
Using tempfile C:\Users\z5843\AppData\Local\Temp\tmp9q_4uo0i.lnk for command line:
cl /Fo3rd\win_msvc\poco\src\Ascii.obj /c 3rd\win_msvc\poco\src\Ascii.cpp -DLCD_WIDTH=800 -DLCD_HEIGHT=480 -DAPP_DEFAULT_FONT=\"default\" -DAPP_THEME=\"default\" -DAPP_RES_ROOT="\"./res\"" -DAPP_DEFAULT_LANGUAGE=\"zh\" -DAPP_DEFAULT_COUNTRY=\"CN\" -DAPP_ROOT="\"D:/AWStudioProjects/Workspace/poco_crash_on_win_repro\"" -DWIN32 -D_WIN32 -DWINDOWS /EHsc -D_CONSOLE /FS /Z7 /utf-8 -DWITH_64BIT_CPU -D_WIN64 -DHAVE_LIBC -DSDL_REAL_API -DSDL_HAPTIC_DISABLED -DSDL_SENSOR_DISABLED -DSDL_JOYSTICK_DISABLED -D__STDC_LIMIT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_CONSTANT_MACROS -D_HAS_EXCEPTIONS=0 -D_HAS_ITERATOR_DEBUGGING=0 -D_ITERATOR_DEBUG_LEVEL=0 -D_SCL_SECURE=0-D_SECURE_SCL=0 -D_SCL_SECURE_NO_WARNINGS -D_CRT_SECURE_NO_WARNINGS -D_CRT_SECURE_NO_DEPRECATE -DTK_ROOT="\"D:\\\\AWStudioProjects\\\\awtk_msvc\"" -DWITH_STATE_ACTIVATED=1 -DWITH_MBEDTLS=1 -DENABLE_CURSOR=1 -DWITH_TEXT_BIDI=1 -DWITH_DATA_READER_WRITER=1 -DWITH_EVENT_RECORDER_PLAYER=1 -DWITH_ASSET_LOADER -DWITH_FS_RES -DWITH_ASSET_LOADER_ZIP -DSTBTT_STATIC -DSTB_IMAGE_STATIC -DWITH_STB_IMAGE -DWITH_VGCANVAS -DWITH_UNICODE_BREAK -DWITH_DESKTOP_STYLE -DWITH_SDL -DHAS_STDIO -DHAVE_STDIO_H -DHAS_GET_TIME_US64 -DHAS_STD_MALLOC -DTK_MAX_MEM_BLOCK_NR=3 -DWITH_RES_TOOLS -DWITH_MAIN_LOOP_CONSOLE=1 -DWITH_IME_PINYIN -DWITH_STB_FONT -DWITH_BITMAP_BGRA -DWITH_NANOVG_SOFT -DWITH_FB_BGRA8888=1 -DWITH_NANOVG_AGGE -DWITH_AWTK=1 -D_DEBUG -DDEBUG /DEBUG /MDd /Od /MDd -D_DEBUG -DDEBUG /DEBUG /Od /ID:\AWStudioProjects\awtk_msvc /ID:\AWStudioProjects\awtk_msvc\src /ID:\AWStudioProjects\awtk_msvc\3rd /ID:\AWStudioProjects\awtk_msvc\src\ext_widgets /ID:\AWStudioProjects\awtk_msvc\src\custom_widgets /ID:\AWStudioProjects\awtk_msvc\3rd\fribidi /ID:\AWStudioProjects\awtk_msvc\3rd\mbedtls\include /ID:\AWStudioProjects\awtk_msvc\3rd\mbedtls\3rdparty\everest\include /ID:\AWStudioProjects\awtk_msvc\3rd\pixman /ID:\AWStudioProjects\awtk_msvc\3rd\cairo /ID:\AWStudioProjects\awtk_msvc\3rd\bgfx\bgfx\include /ID:\AWStudioProjects\awtk_msvc\3rd\bgfx\bx\include /ID:\AWStudioProjects\awtk_msvc\3rd\bgfx\bimg\include /ID:\AWStudioProjects\awtk_msvc\3rd\agge /ID:\AWStudioProjects\awtk_msvc\3rd\agg\include /ID:\AWStudioProjects\awtk_msvc\3rd\SDL\src /ID:\AWStudioProjects\awtk_msvc\3rd\SDL\include /ID:\AWStudioProjects\awtk_msvc\3rd\agge\src /ID:\AWStudioProjects\awtk_msvc\3rd\agge\include /ID:\AWStudioProjects\awtk_msvc\3rd\gpinyin\include /ID:\AWStudioProjects\awtk_msvc\3rd\libunibreak /ID:\AWStudioProjects\awtk_msvc\3rd\gtest\googletest /ID:\AWStudioProjects\awtk_msvc\3rd\gtest\googletest\include /ID:\AWStudioProjects\awtk_msvc\tools /ID:\AWStudioProjects\awtk_msvc\3rd\nanovg /ID:\AWStudioProjects\awtk_msvc\3rd\nanovg\gl /ID:\AWStudioProjects\awtk_msvc\3rd\nanovg\base /ID:\AWStudioProjects\awtk_msvc\3rd\nanovg\agge /ID:\AWStudioProj [... cut ...]
   /I3rd\win_msvc\openssl\include
   /I3rd\win_msvc\poco\include
   /I3rd\win_msvc\openssl\src

Ascii.cpp
3rd\win_msvc\poco\src\Ascii.cpp(21): warning C4273: “CHARACTER_PROPERTIES”: dll 链接不一致
D:\AWStudioProjects\Workspace\poco_crash_on_win_repro\3rd\win_msvc\poco\include\Poco/Ascii.h(120): note: 参见“CHARACTER_PROPERTIES”的前一个定义
3rd\win_msvc\poco\src\Ascii.cpp(21): error C2491: “Poco::Ascii::CHARACTER_PROPERTIES”: 不允许 dllimport 静态数据成员 的定义
scons: *** [3rd\win_msvc\poco\src\Ascii.obj] Error 2
```

Ubuntu上运行就正常，Windows上太多坑了。

closed 2024-10-15 by Tracker647