zlgopen/awtk #907: awtk中如何实现这种黑边白底的字体？

Tracker647, 2024-10-15:
[image]
像这样

xianjimli, 2024-10-20:
这个设置一下背景颜色和字体颜色就可以吧。

Tracker647, 2024-10-20:
主要是字体黑边框，光设置字体颜色只有纯色，黑边框都是针对控件的，不知道有没有针对字体的，看Qt上是用画布实现，不知道在awtk上有没有类似的方法。
[image]

xianjimli, 2024-10-20:
没看太明白

Tracker647, 2024-10-20:
> 这个设置一下背景颜色和字体颜色就可以吧。

有直接的设置案例吗？或者只能用画布代码实现？

Tracker647, 2024-10-20:
> 没看太明白

就是字体的黑边，有什么方法实现吗？

Tracker647, 2024-10-20:
不想用图片值，图片值的话英文字母无法覆盖。

xianjimli, 2024-10-20:
不知道这样行不：先画一个大的作为背景，再画个小的。

Tracker647, 2024-10-20:
不行，并不好对上。
[image]

Tracker647, 2024-10-27:
问了老板，这个用freetype实现，写了个例子显示成功，然后看awtk里面有个font_get_glyph正好能和freetype的FT_Load_Char对标，参考了下[如何绘制旋转文字](https://awtk.zlg.cn/docs/awtk_docs/HowTo/how_to_draw_rotate_text.html)写了个控件例子，但是显示起来效果不行，我不清楚哪里的问题？
code:
```

static bitmap_t* draw_border_text_to_bitmap(widget_t *widget, canvas_t* canvas, wchar_t* str) {
 
  border_text_t* border_text = BORDER_TEXT(widget);
  style_t *style = widget->astyle;
  color_t text_color = color_init(255, 255, 255, 255);  
  color_t border_color = color_init(0, 0, 0, 255);
  
  glyph_t glyph;
  uint32_t x = 0, y = 0, i = 0, img_w = 0, h = 0, ox = 0;
  font_vmetrics_t vm;
  uint32_t* image_data = NULL;
  uint32_t baseline = 0;
  bitmap_t* bitmap = NULL;
  uint32_t font_size = canvas->font_size;
  font_manager_t* fm = canvas->font_manager;
  font_t* font = font_manager_get_font(fm, canvas->font_name, font_size);
  return_value_if_fail(font != NULL, NULL);

  vm = font_get_vmetrics(font, font_size);
  h = vm.ascent - vm.descent;
  return_value_if_fail(h > 0, NULL);

  baseline = vm.ascent;
  for (i = 0; str[i]; i++) {
    return_value_if_fail(font_get_glyph(font, str[i], font_size, &glyph) == RET_OK, NULL);
    img_w += glyph.advance + 1;
  }

  bitmap = bitmap_create_ex(img_w, h, 0, BITMAP_FMT_RGBA8888);
  return_value_if_fail(bitmap != NULL, NULL);

  image_data = (uint32_t*)bitmap_lock_buffer_for_write(bitmap);
  memset(image_data, 0x00, img_w * h * 4);

  int border_size = 2;

  for (i = 0; str[i]; i++) {
    return_value_if_fail(font_get_glyph(font, str[i], font_size, &glyph) == RET_OK, NULL);
    int char_x = ox + glyph.x;
    int char_y = baseline + glyph.y;

    //绘制字体边框
    for(int bx = -border_size; bx <= border_size; ++bx){
      for(int by = -border_size; by <= border_size; ++by){
          if (bx == 0 && by == 0) continue; // 跳过中心点

          for (y = 0; y < glyph.h; y++) {
            for (x = 0; x < glyph.w; x++) {
              int32_t img_x = char_x + x + bx;
              int32_t img_y = char_y + y + by;

              uint32_t* pixel = image_data + img_y * img_w + img_x;
              if(img_x >= 0 && img_x < bitmap->w && img_y >= 0 && img_y < bitmap->h){
                const uint8_t* grey = glyph.data + y * glyph.w + x;
                border_color.rgba.a = *grey;
                *pixel = border_color.color;
              }            
            }
          }
      }
    }

    // 绘制字体
    for (y = 0; y < glyph.h; y++) {
      for (x = 0; x < glyph.w; x++) {
        int32_t img_x = char_x + x;
        int32_t img_y = char_y + y;

        uint32_t* pixel = image_data + img_y * img_w + img_x;
        const uint8_t* grey = glyph.data + y * glyph.w + x;

        text_color.rgba.a = *grey;
        *pixel = text_color.color;
      }
    }
    ox += glyph.advance + 1;
  }
  bitmap_unlock_buffer(bitmap);
  bitmap->flags |= BITMAP_FLAG_CHANGED;

  return bitmap;
}

static ret_t border_text_on_paint_self(widget_t* widget, canvas_t* c) {
  border_text_t* border_text = BORDER_TEXT(widget);
  style_t *style = widget->astyle;
  int font_size = style_get_int(style, STYLE_ID_FONT_SIZE, 25);
  color_t text_color = color_init(255, 255, 255, 255);  
  color_t border_color = color_init(0, 0, 0, 255);
  
  canvas_set_text_align(c, ALIGN_H_CENTER, ALIGN_V_MIDDLE);
  canvas_set_text_color(c, text_color);
  rect_t r = rect_init(0, 0, widget->w, widget->h);
  
  canvas_set_font(c, "default", font_size);
  bitmap_t *bm = draw_border_text_to_bitmap(widget, c, L"Hello World");
  rect_t dst = {0, 0, widget->w, widget->h};
  rect_t src = {0, 0, bm->w, bm->h};
  canvas_draw_image(c, bm, &src, &dst);
  bitmap_destroy(bm);
  return RET_OK;
}

```

Tracker647, 2024-10-27:
freetype的例子：
```
#include <sys/mman.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <unistd.h>
#include <linux/fb.h>
#include <fcntl.h>
#include <stdio.h>
#include <string.h>
#include <math.h>
#include <wchar.h>
#include <sys/ioctl.h>
#include <ft2build.h>
#include FT_FREETYPE_H
#include FT_GLYPH_H

int fd_fb;
struct fb_var_screeninfo var;	/* Current var */
int screen_size;
unsigned char *fbmem;
unsigned int line_width;
unsigned int pixel_width;

/**********************************************************************
 * 函数名称： lcd_put_pixel
 * 功能描述： 在LCD指定位置上输出指定颜色（描点）
 * 输入参数： x坐标，y坐标，颜色
 * 输出参数： 无
 * 返 回 值： 会
 * 修改日期        版本号     修改人	      修改内容
 * -----------------------------------------------
 * 2020/05/12	     V1.0	  zh(angenao)	      创建
 ***********************************************************************/ 
void lcd_put_pixel(int x, int y, unsigned int color)
{
	unsigned char *pen_8 = fbmem+y*line_width+x*pixel_width;
	unsigned short *pen_16;	
	unsigned int *pen_32;	

	unsigned int red, green, blue;	

	pen_16 = (unsigned short *)pen_8;
	pen_32 = (unsigned int *)pen_8;
	
	switch (var.bits_per_pixel)
	{
		case 8:
		{
			*pen_8 = color;
			break;
		}
		case 16:
		{
			/* 565 */
			red   = (color >> 16) & 0xff;
			green = (color >> 8) & 0xff;
			blue  = (color >> 0) & 0xff;
			color = ((red >> 3) << 11) | ((green >> 2) << 5) | (blue >> 3);
			*pen_16 = color;
			break;
		}
		case 32:
		{
			*pen_32 = color;
			break;
		}
		default:
		{
			printf("can't surport %dbpp\n", var.bits_per_pixel);
[... 152 lines cut ...]
    //FT_Set_Transform( face, 0, &pen);

    /* load glyph image into the slot (erase previous one) */
    error = FT_Load_Char( face, chinese_str[0], FT_LOAD_RENDER );
	if (error)
	{
		printf("FT_Load_Char error\n");
		return -1;
	}
	
    // draw_bitmap( &slot->bitmap,
    //              var.xres/2,
    //              var.yres/2, 0x00000000);

	// 使用白色字体和黑色边框
    unsigned int white = 0x00FFFFFF;  // 假设32位颜色模式下白色是0xFFFFFFFF
    unsigned int black = 0x00000000;  // 黑色是0x00000000
    int border_size = 1;              // 边框宽度

   	// 绘制带边框的文字
    draw_text_with_border(face, chinese_str, lcd_x, lcd_y, border_size, white, black);
	return 0;	
}

```

Tracker647, 2024-10-27:
freetype显示效果和awtk对比：
[image]

[image]

Tracker647, 2024-10-27:
> 问了老板，这个用freetype实现，写了个例子显示成功，然后看awtk里面有个font_get_glyph正好能和freetype的FT_Load_Char对标，参考了下[如何绘制旋转文字](https://awtk.zlg.cn/docs/awtk_docs/HowTo/how_to_draw_rotate_text.html)写了个控件例子，但是显示起来效果不行，我不清楚绘制边框的那段代码是哪里的问题？ code:
> 
> ```
> 
> static bitmap_t* draw_border_text_to_bitmap(widget_t *widget, canvas_t* canvas, wchar_t* str) {
>  
>   border_text_t* border_text = BORDER_TEXT(widget);
>   style_t *style = widget->astyle;
>   color_t text_color = color_init(255, 255, 255, 255);  
>   color_t border_color = color_init(0, 0, 0, 255);
>   
>   glyph_t glyph;
>   uint32_t x = 0, y = 0, i = 0, img_w = 0, h = 0, ox = 0;
>   font_vmetrics_t vm;
>   uint32_t* image_data = NULL;
>   uint32_t baseline = 0;
>   bitmap_t* bitmap = NULL;
>   uint32_t font_size = canvas->font_size;
>   font_manager_t* fm = canvas->font_manager;
>   font_t* font = font_manager_get_font(fm, canvas->font_name, font_size);
>   return_value_if_fail(font != NULL, NULL);
> 
>   vm = font_get_vmetrics(font, font_size);
>   h = vm.ascent - vm.descent;
>   return_value_if_fail(h > 0, NULL);
> 
>   baseline = vm.ascent;
>   for (i = 0; str[i]; i++) {
>     return_value_if_fail(font_get_glyph(font, str[i], font_size, &glyph) == RET_OK, NULL);
>     img_w += glyph.advance + 1;
>   }
> 
>   bitmap = bitmap_create_ex(img_w, h, 0, BITMAP_FMT_RGBA8888);
>   return_value_if_fail(bitmap != NULL, NULL);
> 
>   image_data = (uint32_t*)bitmap_lock_buffer_for_write(bitmap);
>   memset(image_data, 0x00, img_w * h * 4);
> 
>   int border_size = 2;
> 
>   for (i = 0; str[i]; i++) {
>     return_value_if_fail(font_get_glyph(font, str[i], font_size, &glyph) == RET_OK, NULL);
>     int char_x = ox + glyph.x;
>     int char_y = baseline + glyph.y;
> 
>     //绘制字体边框
>     for(int bx = -border_size; bx <= border_size; ++bx){
>       for(int by = -border_size; by <= border_size; ++by){
>           if (bx == 0 && by == 0) continue; // 跳过中心点
> 
>           for (y = 0; y < glyph.h; y++) {
>             for (x = 0; x < glyph.w; x++) {
>               int32_t img_x = char_x + x + bx;
>               int32_t img_y = char_y + y + by;
> 
>               uint32_t* pixel = image_data + img_y * img_w + img_x;
>               if(img_x >= 0 && img_x < bitmap->w && img_y >= 0 && img_y < bitmap->h){
>                 const uint8_t* grey = glyph.data + y * glyph.w + x;
>                 border_color.rgba.a = *grey;
>                 *pixel = border_color.color;
>               }            
>             }
>           }
>       }
>     }
> 
>     // 绘制字体
>     for (y = 0; y < glyph.h; y++) {
>       for (x = 0; x < glyph.w; x++) {
>         int32_t img_x = char_x + x;
>         int32_t img_y = char_y + y;
> 
>         uint32_t* pixel = image_data + img_y * img_w + img_x;
>         const uint8_t* grey = glyph.data + y * glyph.w + x;
> 
>         text_color.rgba.a = *grey;
>         *pixel = text_color.color;
>       }
>     }
>     ox += glyph.advance + 1;
>   }
>   bitmap_unlock_buffer(bitmap);
>   bitmap->flags |= BITMAP_FLAG_CHANGED;
> 
>   return bitmap;
> }
> 
> static ret_t border_text_on_paint_self(widget_t* widget, canvas_t* c) {
>   border_text_t* border_text = BORDER_TEXT(widget);
>   style_t *style = widget->astyle;
>   int font_size = style_get_int(style, STYLE_ID_FONT_SIZE, 25);
>   color_t text_color = color_init(255, 255, 255, 255);  
>   color_t border_color = color_init(0, 0, 0, 255);
>   
>   canvas_set_text_align(c, ALIGN_H_CENTER, ALIGN_V_MIDDLE);
>   canvas_set_text_color(c, text_color);
>   rect_t r = rect_init(0, 0, widget->w, widget->h);
>   
>   canvas_set_font(c, "default", font_size);
>   bitmap_t *bm = draw_border_text_to_bitmap(widget, c, L"Hello World");
>   rect_t dst = {0, 0, widget->w, widget->h};
>   rect_t src = {0, 0, bm->w, bm->h};
>   canvas_draw_image(c, bm, &src, &dst);
>   bitmap_destroy(bm);
>   return RET_OK;
> }
> ```

Tracker647, 2024-10-27:
OK, 查了点文章，字体从主体到边框是有灰度的过渡的，a值不一定是255，可以通过看a值是不是255来区分字体或者边框，修改了下，总算能看了，就是锯齿明显，勉强用了。
```
    //绘制字体边框
    for(int bx = -border_size; bx <= border_size; ++bx){
      for(int by = -border_size; by <= border_size; ++by){
        for (y = 0; y < glyph.h; y++) {
          for (x = 0; x < glyph.w; x++) {
            int img_x = char_x + bx + x;
            int img_y = char_y + by + y;

            uint32_t* pixel = image_data + img_y * img_w + img_x;
            const uint8_t* grey = glyph.data + y * glyph.w + x;
            if(*grey > 0){
              border_color.rgba.a = 255;
              *pixel = border_color.color;
            }
          }
        }
      }
    }

    

    // 绘制字体
    for (y = 0; y < glyph.h; y++) {
      for (x = 0; x < glyph.w; x++) {
        int img_x = char_x + x;
        int img_y = char_y + y;

        uint32_t* pixel = image_data + img_y * img_w + img_x; 
        const uint8_t* grey = glyph.data + y * glyph.w + x;

        if(*grey > 0){
          text_color.rgba.a = 255;
          *pixel = text_color.color;
        }
      }
    }
```

Tracker647, 2024-10-27:
最简单暴力方法GET
static ret_t border_text_on_paint_self(widget_t* widget, canvas_t* c) {
  border_text_t* border_text = BORDER_TEXT(widget);
  style_t *style = widget->astyle;
  int font_size = style_get_int(style, STYLE_ID_FONT_SIZE, 25);
  const char *font_name = style_get_str(style, STYLE_ID_FONT_NAME, "default");
  color_t text_color = color_init(255, 255, 255, 255);  
  color_t border_color = color_init(0, 0, 0, 255);
  
  canvas_set_text_align(c, ALIGN_H_CENTER, ALIGN_V_MIDDLE);
  
  canvas_set_font(c, font_name, font_size);
  canvas_set_text_color(c, border_color);
  for(int j = 0; j < 3; j++){
    for(int i = 0; i < 3; i++){
      rect_t r = rect_init(i, j, widget->w, widget->h);
      canvas_draw_text_in_rect(c, widget->text.str, widget->text.size, &r);
    }
  }

  canvas_set_text_color(c, text_color);
  rect_t r = rect_init(1, 1, widget->w, widget->h);
  canvas_draw_text_in_rect(c, widget->text.str, widget->text.size, &r);
  return RET_OK;
}
直接canvas_draw_text_in_rect重复画几遍就行了，结果发现出来的效果边框还比自己用控件拼的更整齐？？奇妙。
[image]

xianjimli, 2024-10-27:
666