zlgopen/awtk #923: 片段式Framebuffer，文字显示有锯齿

dmrasf, 2024-12-31:
这个属于正常现象吗？如果用全显存是正常的

xianjimli, 2025-01-01:
不正常，截图看看。

dmrasf, 2025-01-02:
[image]

这个是用片段式显示的文字

[image]

这个是全显存显示的文字

[image]

这个是lcd那里的函数，参考的stm32f103ze的移植程序

dmrasf, 2025-01-02:
```
#define pixel_to_rgba(p)                                                       \
  { (0xff & ((p >> 11) << 3)), (0xff & ((p >> 5) << 2)), (0xff & (p << 3)) }
```

找到问题了，docs/lcd.md里面72-73行这个宏不对，少了一个0xff，拷贝的是这里的程序

closed 2025-01-02 by dmrasf