zlgopen/awtk #931: 自定义软键盘切换page怎么显示按钮的聚焦状态？

Tracker647, 2025-03-18:
根据https://awtk.zlg.cn/docs/awtk_docs/AWTK_Guide/7.InputMethod.html#_7-6-%E8%87%AA%E5%AE%9A%E4%B9%89%E9%94%AE%E7%9B%98的做法实现了一个键盘页面，实现思路是大写键盘，小写键盘，数字键盘各自分了一个page来实现大小写和输入数字，但是发现切换page的时候焦点只有实际程序焦点的切换，但是焦点样式却没有跟着过来，鼠标移动到这个按钮的时候才会刷新成正常的样式。
我试了下在按键事件里面widget_set_focused和在idle_queue函数里面widget_set_focused都没有用，这个有没有解决的方法？

谢谢。
键盘样式：
default.xml
```
<battery_widget>
  <style name="default">
    <normal font_size="32" text_color="#000000"/>
  </style>
</battery_widget>
<border_text>
  <style name="default">
    <normal font_size="32" text_color="#000000"/>
  </style>
</border_text>
<button>
  <style name="default" bg_color="#FFFFFF" round_radius="6" text_color="#FFFFFF">
    <normal border_color="#C3C3C3" text_color="#151515"/>
    <disable border_color="#D9D9D9" text_color="#BEBEBE"/>
    <focused border_color="#3B9AFC" text_color="#7A7A7A"/>
    <over bg_color="linear-gradient(180deg, #0C80FF 0%, #6AB2FA 100%)" border_color="#00000000" border_width="0"/>
    <pressed bg_color="#007AEB" border_color="#0071EB"/>
  </style>
  <style name="arrow_btn" bg_color="#00000000" border_color="#00000000" font_size="25" round_radius="6" text_color="#577699">
    <normal/>
    <focused border_color="#00FFFF"/>
    <pressed text_color="#00FFFF"/>
  </style>
  <style name="bottom_tab_btn" bg_color="#00000000" border_color="#00000000" font_name="Inter_18pt_Medium" font_size="25" icon="icons_Group_3214" icon_at="left" text_align_h="right" text_color="#00C2FF">
    <normal icon="icons_Group_3588" text_color="#FFFFFF"/>
    <disable_of_checked/>
    <focused border_color="#00FFFF" icon="icons_Group_3588" text_color="#FFFFFF"/>
    <focused_of_checked/>
    <normal_of_checked/>
    <over_of_checked/>
    <pressed/>
    <pressed_of_checked/>
  </style>
  <style name="bottom_tab_btn2" bg_color="#00000000" border_color="#00000000" font_name="Inter_18pt_Medium" font_size="25" icon="menu_btn_gear0" icon_at="left" text_align_h="right" text_color="#00C2FF">
    <normal text_color="#FFFFFF"/>
    <disable_of_checked/>
    <focused icon="menu_btn_gear1"/>
    <focused_of_checked/>
    <normal_of_checked/>
    <over_of_checked/>
    <pressed icon="menu_btn_gear1" y_offset="2"/>
    <pressed_of_checked/>
  </style>
  <style name="combobox_down" bg_color="#FAFAFA" border_color="#C2C2C2" border_width="1" icon="combo_arrow_n">
    <normal border="left"/>
    <focused border="left"/>
    <over bg_color="linear-gradient(180deg, #0C80FF 0%, #6AB2FA 100%)" border_color="#2F8DFC" border_width="0" icon="combo_arrow_o" round_radius_bottom_right="6" round_radius_top_left="0" round_radius_top_right="6"/>
    <pressed bg_color="#007AFF" border_color="#2F8DFC" icon="combo_arrow_p" round_radius_bottom_right="6" round_radius_top_right="6"/>
  </style>
  <style name="confirm_button" bg_color="#11487B7B" font_name="400Inter_Regular" font_size="24" round_radius="0" text_color="#FFFFFF">
    <normal border_color="#0F395F00"/>
    <focused border_color="#00FFFF"/>
    <pressed bg_color="#00C2FF" border_color="#00000000"/>
  </style>
  <style name="empty_btn" bg_color="#00000000" border_color="#00000000" round_radius="0" text_align_h="left">
    <normal/>
    <disable/>
    <focused border_color="#00FFFF"/>
    <over/>
    <pressed/>
  </style>
  <style name="initial_select" bg_color="#11487B" border_color="#00000000" font_name="400Inter_Regular" font_size="24" round_radius="6" text_color="#FFFFFF">
    <normal/>
[... 1167 lines cut ...]
    } else if (tk_str_eq(name, "alphabet_key")) {
      widget_on(widget, EVT_CLICK, back_to_previous_alphabet_panel, win);
    } else if (tk_str_eq(name, "number_mode_key")) {
      widget_on(widget, EVT_CLICK, on_number_mode, win);
    } else if (tk_str_eq(name, "action")) {
      widget_on(widget, EVT_CLICK, on_action_click, win);
    } 
  }

  return RET_OK;
}

/**
 * 初始化窗口
 */
ret_t home_page_init(widget_t* win, void* ctx) {
  (void)ctx;
  return_value_if_fail(win != NULL, RET_BAD_PARAMS);

  widget_foreach(win, visit_init_child, win);

  return RET_OK;
}

```

KiritoZz6, 2025-04-02:
切换界面后焦点是停留在哪里呢？

Tracker647, 2025-05-17:
> 切换界面后焦点是停留在哪里呢？

可以在这个demo复现情况：

[keyboard_focus_test.tar.gz](https://github.com/user-attachments/files/20261221/keyboard_focus_test.tar.gz)
问题：能正常聚焦到指定控件，但是没有显示focus的样式

[image]

需要使用commit hash为3dc2be39458cab70876125dcbd37427e149f8f88及其之前的awtk版本。

ufbycd, 2025-06-07:
你的思路错了，软键盘输入时，焦点只能停留在输入控件上，软键盘上的任何按钮被点击了也一样。
要实现你想要的效果，只需在大小写page上分别使用不同的⇧键样式即可