zlgopen/awtk #934: 在vmware配了Ubuntu，在运行awtk designer 时报错

DongDongProMax, 2025-03-28:
已经给虚拟机配了IDE的硬盘类型，但是依旧提示：错误代码：6（获取系统uuid失败）
[image]

xuchaoze, 2025-04-02:
请执行一下这个命令看看结果是什么？

pkexec dmidecode -s system-uuid

DongDongProMax, 2025-04-02:
> 请执行一下这个命令看看结果是什么？
> 
> pkexec dmidecode -s system-uuid

$ pkexec dmidecode -s system-uuid
f86b4d56-6a13-88c4-04df-0efacede3ef4

xuchaoze, 2025-04-02:
报错对话框的内容可以全部拷贝上来看看吗

DongDongProMax, 2025-04-02:
> 报错对话框的内容可以全部拷贝上来看看吗

错误代码：6（获取系统uuid失败）。

请试试以下办法：
* 虚拟机
  修改虚拟机硬盘的类型为IDE类型
* Windows系统
  把下列路径添加加到系统环境变量Path中后，重新启动程序：
    C:/Windows;
    C:/Windows/system32;
    C:/Windows/System32/Wbem;
    C:/Windows/System32/WindowsPowerShell/v1.0;
  如果仍然失败，请在cmd.exe中执行下列命令，并把结果反馈给我们：
    powershell -Command "(get-wmiobject Win32_ComputerSystemProduct).UUID"
    wmic csproduct get UUID
* Ubuntu系统
  确保系统硬盘有充足的可用空间。
  如果仍然失败，请在终端中执行下列命令，并把结果反馈给我们：
    dmidecode -s system-uuid

record:

xuchaoze, 2025-04-02:
能发一下 ~/.config/AWStudio/aw_license/licence_debug.log 和 ~./config/AWStudio/aw_license/license_debug1.log 这个两个文件来一下吗？或者发到技术交流群（群加入方式：加 "ZLG致远电子" 公众号，发送 awtk）

DongDongProMax, 2025-04-02:
> 能发一下 ~/.config/AWStudio/aw_license/licence_debug.log 和 ~./config/AWStudio/aw_license/license_debug1.log 这个两个文件来一下吗？或者发到技术交流群（群加入方式：加 "ZLG致远电子" 公众号，发送 awtk）

只找到了：~/.config/AWStudio/aw_license/licence_debug.log，并没有找到~./config/AWStudio/aw_license/license_debug1.log
[licence_debug.log](https://github.com/user-attachments/files/19561603/licence_debug.log)

xuchaoze, 2025-04-02:
请问Studio启动后有正常输入系统的root权限密码吗

DongDongProMax, 2025-04-03:
> 请问Studio启动后有正常输入系统的root权限密码吗

有的，是正常输入正确的密码的

DongDongProMax, 2025-04-03:
> 请问Ubuntu是什么版本

版本：ubuntu-24.10-desktop-amd64

xuchaoze, 2025-04-03:
原因是ubuntu24.10上nodejs执行pkexec会有较长的延迟导致验证超时了，如果急用可以先换个低版本ubuntu，比如24.04

DongDongProMax, 2025-04-07:
> 原因是ubuntu24.10上nodejs执行pkexec会有较长的延迟导致验证超时了，如果急用可以先换个低版本ubuntu，比如24.04

好的