zlgopen/awtk #935: 问题请假：多字体显示

ytchw101, 2025-03-31:
如下：按钮上显示一个特殊字符， 因为默认字体中不含有此字符，而宋体中有 
		widget_t* btn_back = button_create(page.par_textedit, 60, 68, BTN_W, BTN_H);
	        widget_set_tr_text(btn_back, "spe1");
		widget_set_style_str(btn_back, "font_name", "宋体");
		widget_set_style_int(btn_back, "font_size", 24);
		widget_set_style_str(btn_back, "text_color", "black");	

在旧的AWTK源码约20年6月 ，能显示出来 ，显示没问题， 而由于某种原因 更新到 新源码（24年10月的） 显示不出来？

除了AWTK源码不一样，其他源码、硬件等都一样， 且 宋体已经被加载到内存， 内存也够    但字符就是显示不出来...

ytchw101, 2025-03-31:
另外再提一个问题，是在Label上文字显示的问题 原先做好的文字显示都OK, 后来把AWTK源码从20年的 （可以显示）  更新到（24年10月的）  居然不显示了，后来找了老半天原因 发现， 应该是字体大小超出控件宽或高  导致不显示..        我觉得这个问题应该按照之前逻辑处理，不管设定大小超不超控件大小  ， 不应什么都不显示

KiritoZz6, 2025-04-02:
我这里用新的awtk源码编译你上面提供的代码片段加载default.ttf字体，测试创建出来的按钮可以正常显示文本。你可以检查一下在AWStudio中是否导入了该字体文件资源，并且命名是否为英文。