zlgopen/awtk #936: 如何通过画布实现圆形头像？

Tracker647, 2025-04-13:
比如github的头像，给的资源图片都是方形的，设想是先画图片，然后画边框的时候把超出圆形的部分擦除，不知道awtk里面有没有例子？

Tracker647, 2025-04-13:
好的，找到思路了，通过clip_path裁剪路径实现，需要把awtk的画布后端改成cairo。
``` c
static ret_t on_paint_circle_image(widget_t* widget, canvas_t* c) {
  bitmap_t img;
  vgcanvas_t* vg = canvas_get_vgcanvas(c);
  rect_t dst = rect_init(0, 0, widget->w, widget->h);
  
  // 加载图片
  image_manager_get_bitmap(image_manager(), "sonar_view1", &img);

  // 保存当前上下文状态
  vgcanvas_save(vg);
  vgcanvas_set_stroke_color(vg, color_init(165, 165, 165, 255));
  vgcanvas_set_line_width(vg, 1);

  vgcanvas_translate(vg, c->ox, c->oy);  

  // 创建圆形裁剪区域
  float_t r = tk_min(widget->w, widget->h) / 2.0;
  float_t cx = widget->w / 2.0;
  float_t cy = widget->h / 2.0;
  
  vgcanvas_begin_path(vg);
  vgcanvas_arc(vg, cx, cy, r, 0, 2 * M_PI, FALSE);
  vgcanvas_clip_path(vg);
  vgcanvas_stroke(vg);
  
  // 绘制图片
  vgcanvas_draw_image(vg, &img, 0, 0, img.w, img.h, 0, 0, widget->w, widget->h);
  
  // 恢复上下文
  vgcanvas_restore(vg);
  
  return RET_OK;
}
```

[image]

scriptforfivem, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

fivemeyestore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemserverhostings, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

closed 2025-05-25 by Tracker647