zlgopen/awtk #939: 让awtk和awtk-linux-fb多点触控支持返回触点数组

Tracker647, 2025-05-15:
在嵌入式linux平台做双指缩放，需要去计算scaleFactor，感觉AWTK自带的那个multi_gesture_event_t面对复杂点的逻辑（比如地图图片两个缩放级别之间的过渡放大）还是不够用，我想直接把tslib的采样数组拉到应用层去做逻辑，除了改multi_gesture_event_t结构（已经是改库了）根本没有外部渠道，这很不合理，像QT的touchEvent就支持用at方法返回对应下标触点数据。
能否在后面的更新中支持在应用层获取触点数组? 比如：

```

/**
 * @class multi_gesture_event_t
 * @annotation ["scriptable"]
 * @parent event_t
 * 多点触摸手势事件。
 */
typedef struct _multi_gesture_event_t {
  event_t e;
  /**
   * @property {xy_t} x
   * @annotation ["readable", "scriptable"]
   * 中心点x坐标。
   */
  xy_t x;
  /**
   * @property {xy_t} y
   * @annotation ["readable", "scriptable"]
   * 中心点y坐标。
   */
  xy_t y;
  /**
   * @property {float_t} rotation
   * @annotation ["readable", "scriptable"]
   * 旋转角度(幅度)增量。（单位弧度）
   */
  float_t rotation;
  /**
   * @property {float_t} distance
   * @annotation ["readable", "scriptable"]
   * 两点间的距离增量。(-1,0)表示缩小，(0-1)表示增加。
   */
  float_t distance;
   /**
   * @property {touch_event_t} touch_event
   * @annotation ["readable", "scriptable"]
   * 五个手指的触点数据
   */
  touch_event_t touch_event[5];
} multi_gesture_event_t;

```

xianjimli, 2025-05-16:
看下：touch_event_t

完整示例：https://github.com/zlgopen/awtk-web/blob/master/examples/AwtkApplicationCTouchEvent/src/pages/home_page.c

Tracker647, 2025-05-17:
好的谢谢

closed 2025-05-17 by Tracker647