zlgopen/awtk #943: 代码质量堪优，master分支老是通不过编译，整个编译警告有100多个

ufbycd, 2025-06-02:
我的GCC：
```
$ gcc -v              
使用内建 specs。
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/lto-wrapper
目标：x86_64-pc-linux-gnu
配置为：/build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++,rust --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
线程模型：posix
支持的 LTO 压缩算法：zlib zstd
gcc 版本 14.2.1 20250207 (GCC)
```

master分支之前出现需要手动修改源码才能通过编译的情况，现在还是这样：
```
src/widgets/combo_box.c: In function 'combo_box_on_key_event':
src/widgets/combo_box.c:450:35: error: passing argument 2 of 'edit_on_event' from incompatible pointer type [-Wincompatible-pointer-types]
  450 |       ret = edit_on_event(widget, evt);
      |                                   ^~~
      |                                   |
      |                                   key_event_t * {aka struct _key_event_t *}
In file included from src/widgets/combo_box.h:25,
                 from src/widgets/combo_box.c:31:
src/widgets/edit.h:701:48: note: expected 'event_t *' {aka 'struct _event_t *'} but argument is of type 'key_event_t *' {aka 'struct _key_event_t *'}
  701 | ret_t edit_on_event(widget_t* widget, event_t* e);
```
并且多少年了，编译警告一直维持100多个。
```
13:31:59 Build Finished. 0 errors, 120 warnings. (took 45s.615ms)
```

xianjimli, 2025-06-04:
这个编译错误，在gcc11/clang/vc 中都是警告，也不会影响逻辑。

xianjimli, 2025-06-05:
是的，谢谢指出。有时候想偷懒，需要人推一把。

closed 2025-06-05 by ufbycd